mysensors/NodeManager #50: Heartbeat for timer

Shagrat2, 2017-04-09:
Append Heartbeat for timer

user2684, 2017-04-09:
Hi, what do you mean exactly? A hearbeat is sent automatically when waking up from sleep. Do you mean to do something similar also after a wait cycle?

Shagrat2, 2017-04-09:
For the battery - if you do not send data (sensors data no change - is set setTackLastValue)
For no sleep - Does not have battery status and sensors (electronic lock or sprinkler) - send "I'm alive"

user2684, 2017-04-09:
Ok, so for non sleeping nodes, battery status reports are sent if the node is in a WAIT cycle (wait or sleep is handled in the same way by NodeManager). You are saying, what about a node which is not sleeping AND not using wait. True, they will not send out battery level notifications. BUT I'd recommend in that case to set a sleep (WAIT) cycle anyway. E.g. 1 day wait cycle and battery reports every 1 cycle. Actuators (relay, switch, etc.) do nothing during loop() so it is safe to set a sleeping cycle and when needed you can interact with them by sending a message. Does it make sense? You have a point though since when using wait you can get a battery level but you don't get any heartbeat. I've opened #58 for this

To your other point, what if setTackLastValue() is set and the data does not change. You should get battery level reports in this case even if the value does not change since the battery report routine in the code is called regardless if the sensor is reporting or not any data.

Shagrat2, 2017-04-10:
My node not have battery. What status to be sent?

user2684, 2017-04-10:
ok, without battery you then need to wait for #58 so to have a heartbeat sent out periodically. Is this what you needed right?

Shagrat2, 2017-04-11:
Yes

```
// setup
void setup() { 
  wdt_enable(WDTO_8S);    
  lastSend=millis();    
}

// loop
void loop() {
  currentTime = millis();

  // call NodeManager loop routine
  nodeManager.loop(); 

  // Send feq
  if (currentTime - lastSend > SEND_FREQUENCY){
    sendHeartbeat();     
    lastSend=currentTime;
  }
} 
```

user2684, 2017-04-11:
Ok perfect, this is exactly what I had in mind even if the implementation will be slightly different (with a WAIT cycle). I'll merge #58 here then since they are referring to the same thing. Keep on eye on this issue and ignore #58 then.

closed 2017-04-12 by user2684