mysensors/NodeManager #55: Add support for ACS712

bilbolodz, 2017-04-09:


user2684, 2017-04-09:
I don't have this sensor so instead of waiting a couple of months before being delivered I think it will be quicker to just ask you a few question :)
Is it this something similar to this one? 
* https://www.aliexpress.com/store/product/NEW-20A-Hall-Current-Sensor-Module-ACS712-model-20A-deal-in-all-kind-of-electrocnic-components/110055_32664534976.html?spm=2114.12010615.0.0.lYeXaa

If so it is just a matter of handling the analog output which makes thing extremely easier because I can put together the code and share it with you to test if you are fine with it

bilbolodz, 2017-04-09:
Exactly. "From Arduino point of view" it's just analog voltage sensor. Look at:
[http://henrysbench.capnfatz.com/henrys-bench/arduino-current-measurements/the-acs712-current-sensor-with-an-arduino/](url)

`/*
Measuring Current Using ACS712
*/
const int analogIn = A0;
int mVperAmp = 185; // use 100 for 20A Module and 66 for 30A Module
int RawValue= 0;
int ACSoffset = 2500; 
double Voltage = 0;
double Amps = 0;

void setup(){ 
 Serial.begin(9600);
}

void loop(){
 
 RawValue = analogRead(analogIn);
 Voltage = (RawValue / 1024.0) * 5000; // Gets you mV
 Amps = ((Voltage - ACSoffset) / mVperAmp);
 
 
 Serial.print("Raw Value = " ); // shows pre-scaled value 
 Serial.print(RawValue); 
 Serial.print("\t mV = "); // shows the voltage measured 
 Serial.print(Voltage,3); // the '3' after voltage allows you to display 3 digits after decimal point
 Serial.print("\t Amps = "); // shows the voltage measured 
 Serial.println(Amps,3); // the '3' after voltage allows you to display 3 digits after decimal point
 delay(2500); 
 
}`

bilbolodz, 2017-04-09:
There are 3 versions 05, 20, 30 number means max current.

user2684, 2017-04-09:
Cool, I'll work on it as the first thing for v1.5 and I'll let you know when ready to test

user2684, 2017-05-07:
Added SENSOR_ACS712 depending on MODULE_ANALOG_INPUT.
Added sensor-specific functions:
```
    // set how many mV are equivalent to 1 Amp. The value depends on the module (100 for 20A Module, 66 for 30A Module) (default: 185);
    void setmVPerAmp(int value);
    // set ACS offset (default: 2500);
    void setOffset(int value);
```
This will be made available in v1.5-dev2 which will be pushed to github later today

closed 2017-05-07 by user2684