mysensors/NodeManager #72: Optimize battery life with DS18B20

Toshik, 2017-05-01:
For now NodeManager uses "standard" DS18B20 data retrieval:
```
  _sensors->requestTemperatures();
  float temperature = _sensors->getTempCByIndex(_index);
```
So, requestTemperatures() uses just delay/loop to wait required time for data conversion from DS18B20. And this time is up to 750ms

My proposal is use async mode, as I do in my battery powered projects: 
disable delay in requestTemperatures() by calling ```_sensors->setWaitForConversion(false)```. Then save some power by:
```
  _sensors->requestTemperatures();
  sleep(750); // or calculate exact wait time according to current DS18B20 resolution
  float temperature = _sensors->getTempCByIndex(_index);
```

That way we can make battery work little bit longer

user2684, 2017-05-02:
Thanks for reporting this! I'll make it as an option for the next release

luizrrocha, 2017-05-02:
Hello Toshik/user2684.

I do like this:
  // Fetch temperatures from Dallas sensors
  sensors.requestTemperatures();

  // query conversion time and sleep until conversion completed
  int16_t conversionTime = sensors.millisToWaitForConversion(sensors.getResolution());
  sleep(conversionTime);

This way I do not have to worry about finding the sleep time :-)

user2684, 2017-05-07:
I've added `void setSleepDuringConversion(bool value);` to the SensorDs18b20 class, default is false to avoid creating issues with non-sleeping nodes.

For those using the default DallasTemperature library (not the one in the MySensors bundle), millisToWaitForConversion() is declared as private so has to be moved into public in DallasTemperature.h to have this compiling correctly.
The update will be made available in v1.5-dev2 that will be pushed to github later today.

closed 2017-05-07 by user2684