mysensors/NodeManager #73: Reversed SET/REQ logic in case of OUTPUT type sensors

bilbolodz, 2017-05-04:
In case of output sensors set command should "set" new value of sensor and req command should return current state of sensor

user2684, 2017-05-05:
I'll add an option to set which command to expect rather than changing the default behavior since I'm afraid this could break compatibility with previous version.

bilbolodz, 2017-05-05:
Fine, and please add option which to define which state (LOW/HIGH) means for relay open/close. Depends on hardware (especially when using board with photocoupler) some relay needs low signal to fire.

user2684, 2017-05-05:
Good one, I've added #77 for the latter. Thanks!

user2684, 2017-05-06:
Not sure you see the updates on #77 but please have a look at my comment there and let me know if my understanding is correct. Thanks

user2684, 2017-05-07:
I have implemented it in a way closer to what you suggested rather than my proposal. Reason is I should already know which sensors are output sensors so it does not make sense to ask the user to do the job. Also onReceive() should be able to handle both C_REQ and C_SET command for those sensors hence I've done the following changes:
* removed the filter on NodeManager's receive() for C_REQ. Now all the messages (both C_REQ and C_SET) will be passed along the sensor's onReceive()
* modified SensorDigitalOutput class to handle C_REQ and C_SET differently. C_SET will now set the new value, C_REQ will return the state as requested. This of course applies also to the derived class SensorRelay and SensorLatchingRelay. 
* For the other out of the box input sensors I've added the filter for C_REQ
* generally speaking it will be up to the sensor-specific implementation of onReceive() to handle the two commands differently if needed

This will break compatibility with v1.4 and previous version of NodeManager expecting C_REQ instead of C_SET but looks like that first implementation was wrong so I feel like this is the way to go. I'll try to make this clear in the release notes.
This change will be available in v1.5-dev2 that will be pushed to github later today.
Thanks

closed 2017-05-07 by user2684