mysensors/NodeManager #77: Add option to define which state means for relay open/close

user2684, 2017-05-05:


user2684, 2017-05-06:
I've just realized there is already a setInitialValue() default to LOW in the SensorRelay class so it should be already possible to achieve it while when receiving a user's request the payload tells which state to set. 
Let me know if I have misunderstood something

bilbolodz, 2017-05-06:
I've tried to play with it. Default value it's different issue that state for open/close. State tells how to drive relay to switch it on. Default value tells: is relay should be on or off in "default state". Both functionality are needed.

user2684, 2017-05-07:
Ok something like a setOnValue() function which will set a variable which by default is set to HIGH but can be set to LOW by the user so when the user sends a message with a payload equals to 1, instead of using it in the digitalWrite(), LOW will be used instead. Right?

bilbolodz, 2017-05-08:
I think it's better do it in "constructor". Assigning these value should be done ASAP to prevent sending "not correct" signal to relay and accidentally not turn relay on (if not needed) during startup. Generally: first you should set needed "default value" and then enable Arduino output.

user2684, 2017-05-08:
Having this in the constructor would be very complicated in the actual implementtion since when registering a sensor you are not allowed to pass other parameters. However, the initialization is done in before() which should be early enough I hope. Have a look at the implementation in the commit here above and in case you think it will not work let me know and I will re-open this.
Thanks!

closed 2017-05-08 by user2684

user2684, 2017-05-22:
Added the following for compatibility with older versions:
```
    // when legacy mode is enabled expect a REQ message to trigger, otherwise the default SET (default: false)
    void setLegacyMode(bool value);
```

commit d296022b (2017-05-08): Add option to define which state means for relay open/close #77