mysensors/NodeManager #81: New sensor type - AC main present

bilbolodz, 2017-05-07:
It could be a good idea to add sensor which detect presence of AC mains. It could be implemented by putting AC optocoupler, pooling state and checking if "0V" state last not to long (it should change from 0 to "1" at least 100 times per second).

user2684, 2017-05-07:
Hi, forgive me lack of knowledge, can #55 do the job or is it something completely different?
Thanks

bilbolodz, 2017-05-08:
Yes and now. In some cases you don't need to measure current (it's not necessary or it's too big) but it's good to know if there is a AC main present. Sometimes (in critical situations) it's better not to put measuring element in series in load (if ACS712 fries could be a problem). Putting detection element in parallel with load do the job in safer way (e.x. detect 3 phase on AC on power of whole house; putting 3xACS712 - I'm not so brave, but connect in parallel 3 small optocouplers -why not). I think it could a good improvement to your code.

user2684, 2017-05-08:
Oh I see, thanks for the explanation. Do you have by chance some pointers to share with me or even where I can find some sample code? I found this https://www.fairchildsemi.com/datasheets/MI/MID400.pdf but not sure I'm looking into the right direction.

Thanks!

bilbolodz, 2017-05-09:
Looks fine. Some therory/ideas you can find here: 
http://playground.arduino.cc/Main/ACPhaseControl
http://www.instructables.com/id/Arduino-controlled-light-dimmer-The-circuit/

because it's only detection (not driving) maybe it's enough to use pooling (rather then interrupt) way of reading state (in "ordinary" Arduino there is very limited number of interrupt capable lines).

user2684, 2017-05-09:
Thanks! I'll definitely try to implement something for v1.6 and come back to you in case of any doubts

user2684, 2018-02-25:
Hi, I'm finally working on this but I got lost almost immediately (lack of knowledge :p). I'm aiming at detection only for now and I started from http://playground.arduino.cc/Main/ACPhaseControl. I got lost with all the numbers and how to select the variables. Also, do I need to use the GATE or since it is detection only it doesn't apply? I'm also unsure which kind of mysensors' type and presentation to use for such a sensor. Last concern is about the timing: using such a code with mysensors library and nodemanager, is this going to work? (considering it has to measure delay of a few milliseconds and the library can introduce a much larger overhead). Bottom line, in case you have any sample code to help me with would be great :) Thanks!