mysensors/NodeManager #83: Use time library to keep track of the current time after requesting it

user2684, 2017-05-07:
http://playground.arduino.cc/Code/time

getlarge, 2017-05-30:
Any progress on this one? You want to save time on a RTC ?
Based upon the time aware sensor sketch, i made a routine in which the time of the controller is transmitted to the node at the init. When the message is received, it synchronizes with internal timer of the arduino and with the RTC. And in case time from controller can't be received, we use the RTC time.
This part was related to the scheduler i told you about.

user2684, 2017-05-30:
This will be pretty complex to put in place, I'm still figuring out the most simple and effective way. Good idea to use a RTC module, it would avoid asking the controller continuously. 
I'm also looking for a library which can be installed if possible directly from the arduino ide so to be consistent with the other dependencies. Thanks

jsiddall, 2018-01-25:
I will also vote for this one.  I am not even that interested in using an RTC, just in being able to keep track of the ToD.  All that should be required for that is a periodic, and ideally configurable (ex: hourly), call to requestTime() and then simply calling setTime(ts) to receiveTime() to keep the clock in sync.

I believe the crystal in the Arduinos has a tolerance of 0.5% which is less than a minute per hour of drift.

user2684, 2018-01-27:
I've started working on it recently, keep on eye on #259 while it progresses. Thanks

PR #259 (2018-02-04): Time Aware capability
This PR is to optionally allow the node to keep track of the current time, either with an attached RTC or by requesting the time to the controller.

* Added time support through the Arduino's time library when FEATURE_TIME is turned on (https://github.com/PaulStoffregen/Time). Fixes #83, #84
* Automatically synchronize the local clock 1) during setup() 2) during loop() if last updated more than 60 minutes ago, 3) after waking up from sleep
* Added RTC support (DS3232 and DS3231 through https://github.com/JChristensen/DS3232RTC) when FEATURE_RTC is on. RTC time is always s [... cut]
closed 2018-02-04 by user2684