mysensors/NodeManager #94: Add rule-based engine with actions triggered based on configured condition

user2684, 2017-05-14:


getlarge, 2017-06-26:
You want to handle this on node side ? gateway ?
We've been working on this but that's a job for our controller... we called these : scenarios. Easier to configure for the user, and the dev ;)
We also thought about making a copy of created rules to the fs of the gateway. But I'm not sure one ESP8266 as a gateway could handle this job without the risk of dropping entering message when it acts as a conductor...

user2684, 2017-06-26:
I fully agree with you. I was just thinking of basic rules to allow registered sensors interacting each other when needed. Something like temperature below a threshold, then turn a relay and a heater on. But I'd probably keep this on-hold since it is a lot of effort and probably, as you said, not extremely useful since it is more the job of the controller

user2684, 2018-03-31:
Closing this since can be implemented through #312 by registering a hooking function

closed 2018-03-31 by user2684