mysensors/NodeManager #95: Add capability to show data on a LCD

user2684, 2017-05-14:
This would depend in #93 with a global lcd object in NM and the other sensors accessing it plus a global routine

bilbolodz, 2017-05-14:
Maybe better is to use i2c connected LCD and implement "general use" something like MySensors to I2C bridge? Just idea maybe stupid.

user2684, 2017-05-14:
It is a good idea indeed, I should also have a i2d LCD somewhere to test. I need to structure a good generic use case so to have NM managing the LCD without the user to necessarily do it in a different way for every sensor. Thanks

user2684, 2017-05-31:
Also add a S_INFO child so text can be set also remotely

user2684, 2018-02-11:
This feature must be LCD-related, closing it but referring implementation to #281 and #280

closed 2018-02-11 by user2684