mysensors/NodeManager #96: Pulse Sensor

Shagrat2, 2017-05-15:
Append pulse sensor

Power Meter Pulse Sensor
Water Meter Pulse Sensor

user2684, 2017-05-15:
Good one!
I'll take https://www.mysensors.org/build/pulse_power and https://www.mysensors.org/build/pulse_water as examples to build those sensors into NodeManager. Thanks

user2684, 2017-05-18:
Bought the following, waiting for delivery:
* https://www.aliexpress.com/item/Photosensitive-Sensor-Module-Light-Detection-Module-for-Arduino/1886288541.html?spm=2114.13010608.0.0.pVTs5l
* https://www.aliexpress.com/item/free-shipping-TCRT5000-infrared-reflectance-sensor-Obstacle-avoidance-module-tracing-sensor-tracing-module/32214044656.html?spm=2114.13010608.0.0.pVTs5l

user2684, 2017-07-30:
Done, implementation details on #194

PR #194 (2017-07-30): Support for Power/Meter pulse sensors
Fixes #96 

* Since both power meter and water meter sensors work with a logic similar to SensorRainGauge, this has been generalized into a SensorPulseMeter class 
* SENSOR_RAIN_GAUGE now depends on MODULE_PULSE and setSingleTip() has been converted into setPulseFactor() where pulse_factor = 1 / single_tip
* SENSOR_POWER_METER and SENSOR_WATER_METER have been added depending on MODULE_PULSE
* Added setPulseFactor() and setInterruptMode() to SensorPulseMeter  class
* Added SensorPowerMeter and SensorWaterMeter classes with reasonable default values
* Since  [... cut]
closed 2017-07-30 by user2684