mysensors/NodeManager #98: Changing node id remotely doesn't work

user2684, 2017-05-16:
I had to remove the include of MyHwAVR.h since creating problems between versions hence hwWriteConfig() cannot be called

getlarge, 2017-05-30:
That might be related to Mysensors lib update? Hardware files have been removed to hal directory.

user2684, 2017-05-30:
Correct, I used to include MyHwAVR.h to execute hwWriteConfig() but I preferred to avoid it even if it is still there in the HAL directory now otherwise I need to find a way to include a different file based on the lib version and it is quite a mess. Additionally if possible I prefer not using internal functions which can be changed at any time. Still I need to figure out how to solve it in a different way :)

user2684, 2018-02-11:
Closing this since I don't think it would be reliable to change the node id on the fly and would require hw-specific hwWriteConfig() to be called (https://github.com/mysensors/MySensors/blob/af3452f91c15b60cc7666b6225fa7342f08a6f52/core/MyTransport.cpp#L126)

closed 2018-02-11 by user2684