mysensors/NodeManager #103: Add hook function for before/after set sensor (e.x. relay) call

bilbolodz, 2017-05-22:
I don't know if it's supported in Adruino but maybe using weak functions?
I've 4 relays and ONLY one (of four) can be enabled in the same time. The best solution for me is to add hook function BEFORE turning on relay to be 100% sure that other relays are off. Also hook after setting sensor (relay) could be useful (e.x. to be sure that some effects appear after action).

user2684, 2017-05-23:
This is interesting even if a bit beyond my programming skills ;-) 
Providing different hooks would definitely enhance the flexibility!
I'll look into it 

For my own reference: http://www.keil.com/support/man/docs/armcc/armcc_chr1359124970859.htm

bilbolodz, 2017-05-23:
http://www.keil.com/support/man/docs/armcc/armcc_chr1359124970859.htm

bilbolodz, 2017-05-23:
It's probably NOT implemented into Arduino.

user2684, 2018-02-11:
Hi, I couldn't find a way to implement this, __weak is not supported and even if so I'd be afraid to consume additional storage to hold all the pointers to the hooked functions. If you don't see better alternatives I'll close this for now. 
For your use case above, a workaround could be to define a new custom class derived from SensorRelay and override the function for which you need a different behavior. Thanks!

closed 2018-02-11 by user2684