mysensors/NodeManager #113: Provide a way to program an output sensor remotely (e.g. scheduler)

user2684, 2017-05-27:
> You would configure, from a web interface :
> 
> a scheduler containing : ( id - days to be active - hour - min - seconds )
> it would be sent as a payload to the gateway then to the specified node which will save this scheduler by his id.
> it could stay for general purpose or more oriented to a node relay action, in this case, the message could also contain sensorId ( relay number), desired state/value.
> In the method I tried previously ( with TimeAlarms library), the scheduler message is sent to a specific sensorId ( like your 200 ). When the message is received by the node, it goes through something like this :
> if (sscanf(str, "%d-%s -%d-%d-%d-%d-%d", &alarmId, weekDay, &alarmHour, &alarmMin, &alarmSec, &relayId, &relayState) == 7) {
> And with a Switch-Case, using alarmId number, a routine gets executed to parse the message and define the scheduler internally.
> 
> As I told you before, it's still messy, a bit heavy and even the approach might not be the best. Maybe it should be a periodic reminder with a timestamp of the future action desired for the day ?
> Anyway, I can't build this thing alone !

user2684, 2017-05-29:
A simplified version can take out of the REQ message the time it is supposed to stay on

getlarge, 2017-05-30:
What are you calling REQ message ?

user2684, 2017-05-30:
Sorry, I meant a SET message, depending on how setLegacyMode() is set. I was thinking of a first step before getting to the full scheduler which is pretty complex in which instead of sending out a message with 1 or 0, you send a timeframe (e.g. 10 minutes) and the digital output is automatically turned off after that elapsed time.

user2684, 2017-05-31:
With the latest merge, the behavior I've described above has been implemented (details on #131). I'll keep this open since having a fully features scheduler is still something very interesting and useful but I'll postpone it to future releases since it is a lot of work, hopping mean while #131 can be helpful :)