mysensors/NodeManager #118: Add support for generic PWM output node

rakeshpai, 2017-05-30:
...so that we can dim a light via NodeManager. :)

user2684, 2017-05-30:
That's a good idea. Any sample sketch around I can take inspiration from? Thanks

rakeshpai, 2017-05-30:
w00t! Thanks for considering this.

There are two examples in the MySensors examples set itself. A simple one [is here](https://github.com/mysensors/MySensors/blob/development/examples/DimmableLEDActuator/DimmableLEDActuator.ino). A second fancy one [is here](https://github.com/mysensors/MySensors/blob/development/examples/DimmableLight/DimmableLight.ino).

user2684, 2017-05-30:
Cool thanks! Simple and clean example

rakeshpai, 2017-06-03:
If I may add-on to this issue, could you also consider the ability to enable smooth transitions between levels when a level change is requested. The time for the transition could be configurable (default say 0.5s). 

What would be really awesome is if the transition function could be configurable too. [Here's a list of transition functions](http://easings.net/) commonly used in animations. Of course, we don't need to implement most of these functions. I would recommend just the first 4: Linear, easeInSine, easeOutSine, easeInOutSine. The ease* functions would really help in making the transitions feel natural, rather than the robotic linear transition function.

user2684, 2017-06-03:
Cool idea, noted! Thanks

user2684, 2017-07-29:
For my own reference: https://github.com/chicoplusplus/arduino/tree/master/libraries/Easing

user2684, 2017-07-29:
Added SensorDimmer class, requiring MODULE_DIMMER which would unlock SENSOR_DIMMER. 
Implementation details in #192

PR #192 (2017-07-29): Added support for PWM output
Fixes #118 

* Added MODULE_DIMMER and SENSOR_DIMMER
* Added a single child presenting as S_DIMMER and accepting V_PERCENTAGE
* V_STATUS is not supported (would require an additional child; now the output can be turned off with V_PERCENTAGE = 0. It can be an enhancement of a future release)
* Four transition effects have been implemented: EASE_LINEAR, EASE_INSINE, EASE_OUTSINE,   EASE_INOUTSINE
* User can set the overall transition duration and the duration of each step
* All functions are also available through the remote API

```
    // [101] set the effect to us [... cut]
closed 2017-07-29 by user2684