mysensors/NodeManager #155: Compiler warnings

rakeshpai, 2017-06-27:
I'm using [platformio](http://platformio.org/) as the tool to do compilation, and I get the following warnings:

```
In file included from src/NodeManager.cpp:5:0:
src/NodeManager.h:388:26: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
char* _description = "";
^
src/NodeManager.h:400:28: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
char * _value_string = "";
^
src/NodeManager.h:403:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
char * _last_value_string = "";
^
src/NodeManager.cpp: In member function 'void NodeManager::unRegisterSensor(int)':
src/NodeManager.cpp:2119:26: warning: statement has no effect [-Wunused-value]
_sensors[sensor_index] == 0;
^
src/NodeManager.cpp: In member function 'int NodeManager::_getAvailableChildId()':
src/NodeManager.cpp:2618:1: warning: control reaches end of non-void function [-Wreturn-type]
}
^
Compiling .pioenvs/pro8MHzatmega328/FrameworkArduino/HardwareSerial0.o
In file included from /home/rakesh/Downloads/MyNode/MyNode/src/MyNode.ino:7:0:
src/NodeManager.h:388:26: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
char* _description = "";
^
src/NodeManager.h:400:28: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
char * _value_string = "";
^
src/NodeManager.h:403:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
char * _last_value_string = "";
^
```

Not the end of the world - everything compiles successfully, but I thought I'd bring this to your attention.

If you aren't familiar with platformio, it's an IDE that's been gaining popularity recently. It's most compelling features include dependency management and IDE integration with Atom. The reason I'm interested in it is because it has a CLI, which is useful to perform headless automated dependency resolution, and compilation, which you can't do through the regular Arduino IDE.

user2684, 2017-07-01:
Thanks for letting me know! I guess I don't see those warnings because of different compiler's flags but I'll take care they will go away in the next release :)

user2684, 2017-07-15:
Hi, this should be fixed, if without any urgency you have the time to check https://github.com/user2684/NodeManager/tree/fix/compiler_warnings out would be great :)
Thanks

rakeshpai, 2017-07-18:
Thanks!

However, I now get the following errors:

```
src/NodeManager.cpp: In member function 'virtual void Sensor::loop(const MyMessage&)':
src/NodeManager.cpp:419:21: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
_value_string = "";
^
src/NodeManager.cpp: In member function 'int NodeManager::registerSensor(Sensor*)':
src/NodeManager.cpp:3181:43: warning: return-statement with no value, in function returning 'int' [-fpermissive]
if (sensor->getChildId() > MAX_SENSORS) return;
^
src/NodeManager.cpp: In member function 'void NodeManager::unRegisterSensor(int)':
src/NodeManager.cpp:3206:59: warning: statement has no effect [-Wunused-value]
if (_sensors[sensor_index] != 0) _sensors[sensor_index] == 0;
^
src/NodeManager.cpp: In member function 'bool NodeManager::renameSensor(int, int)':
src/NodeManager.cpp:3221:65: warning: return-statement with no value, in function returning 'bool' [-fpermissive]
if (old_child_id > MAX_SENSORS || new_child_id > MAX_SENSORS) return;
^
```

Not sure how big of a deal these are. I only get them when I do a build after a clean, but not on subsequent builds. These warnings also seem to be unrelated to the original warnings I posted about, not sure.

user2684, 2017-07-22:
Thanks, I probably need to give platformio a try to fix this properly these issues :)
I guess the reason why the warnings are not there on subsequent builds can be because the object is already compiled so it is just linked with the remainder objects. I'll keep you posted.
Thanks

rakeshpai, 2017-07-22:
:+1: If you don't want to install the whole IDE, you could use [their CLI tool](http://docs.platformio.org/en/latest/installation.html) instead.

user2684, 2017-07-29:
Postponing to v1.7 for now :)

graham-hanson, 2017-09-18:
Hi @user2684,

You can enable compiler warnings in the Arduino IDE in File, Preferences - set Compiler Warnings to All.

You should fix the two "return-statement with no value" warnings - in these cases, the functions will return an undefined value (what the C++ standard calls "undefined behaviour" - meaning that you may sometimes get the expected effect but you may get a different effect each time you call the function and that includes sometime crashing the program completely!)

The problem with the line

    if (_sensors[sensor_index] != 0) _sensors[sensor_index] == 0;

which was originally just 

    _sensors[sensor_index] == 0;

is that it should simply be 

    _sensors[sensor_index] = nullptr; // Reset the pointer

i.e., an assignment rather than an equality test - very common mistake but it indicates the importance of using the highest level of warnings from the compiler!
(Note C++11 prefers use of the nullptr literal to zero as it can only be assigned to a pointer variable and is self documenting.)

Regards
Graham (also graham86 on MySensors forum)

user2684, 2017-09-24:
Thanks Graham, I completely missed that option, thanks for the hit!

SergioRius, 2017-10-01:
Just a question. Those "deprecated" warnings, where referring to "(int)myvar" style casts? Because if true, those are C-style casts, being osed on cpp files, and they're old, but completely supported (and btw used by everyone). I also get those warnings in my IDE, but can be safely ignored.

graham-hanson, 2017-10-02:
Hi Sergio,
The deprecated conversion warnings are because the code is assigning a string constant - `""` to a `char *` variable.
The problem is that a string constant has type `const char[]` so assigning to `char *` looses the `const`. (And also because the code could then attempt to write to the constant string, which, probably doesn't have enough space, may be shared with other pieces of code and may be in Read Only memory)
 
In this case, as all the member variables (3 of them from memory) causing this warning are assigned pointers to strings that are never updated, the variables should be declared as e.g.,
    const char * _value_string = "";
(To clarify, the strings assigned to the variables are never updated through the variable, but pointers to different strings are assigned to the variables.)

I have never come across as case where the compiler generates a warning for simply using a C-style cast in a C++ program instead of the C++ equivalent cast.  GCC will warning about certain conversions such as using `(char*)` to remove the constness from a `const char*`,  because it is dangerous.
The use of C-Style casts in a C++ program are not recommended as they, as in the example above, override some of the C++ language protections and result in "dangerous" code.

Finally, it is considered bad practice to ignore compiler warnings, even if .  Best practice is considered to be to compile with the highest warning level available and to "fix" all warnings.  In fact, the preferred method is to enable the compiler "Warnings are Errors" option (-Werror for the gcc compiler used for Arduino).  In practice, it can be difficult or impossible to remove all warnings, in particular because some my be generated by library code. However, the aim should be to reduce the number of warnings as far as possible, and for warnings in your "own" code to be reduced to zero.

user2684, 2018-01-01:
This should be solved with v1.7-dev, especially PRs #241 and #242

closed 2018-01-01 by user2684