mysensors/NodeManager #160: Reverse polarity of sensors that can get wet

rakeshpai, 2017-07-02:
Soil moisture sensors and rain sensors can corrode over time due to the unidirectional DC current flowing through them. (This is due to electrolysis-like effects on the exposed metalwork, I guess?) One way to avoid the DC corrosion is to frequently reverse the polarity of the sensors, say for every alternate reading. Since these are resistive sensors, their reading isn't affected by polarity, and the frequent reversing of polarity extends the life of the sensor.

Polarity can be reversed by powering the sensor using two pins (as NodeManager does now), but alternating the pins' role as source or sink. The ADC reading will have to be adjusted accordingly.

You could use [the example in the MySensors project](https://github.com/mysensors/MySensors/blob/master/examples/SoilMoistSensor/SoilMoistSensor.ino) for reference. It goes through quite some effort to handle this corrosion issue.

user2684, 2017-07-02:
Love this idea!

user2684, 2018-02-11:
Hi, I was reviewing this issue and if not mistaken, this would require a different soil moisture sensor than the one currently supported in NodeManager (https://www.aliexpress.com/item/New-Soil-moisture-meter-to-detect-soil-moisture-sensor-module-robot-smart-car-1pcs/32331823043.html). If so, the idea is still great but its implementation will probably be beyond v1.7 since would require a new Sensor class. Let me know if I'm missing something. Thanks!