mysensors/NodeManager #164: Feature request - extend configuration service with user parameters

stefaanv, 2017-07-03:
The configuration service of NodeManager is a good starting point for better configurability of the sensor nodes.  Adding the possibility to set and update (user-defined) parameters would be a great addition to this.  
My recent post on https://forum.mysensors.org/topic/4970/setting-parameters-in-nodes/9 maybe can serve as an inspiration source.  Having these user parameters proves very usefull.
Still better would be to automatically persist the parameters in eeprom memory after they have been changed.

user2684, 2017-07-05:
Hi, thanks for input! Do you think #102 which part of the development branch only can cover this requirement?
Thanks

stefaanv, 2017-07-11:
Hmmm, not sure the title of #102 sounds about right, but there's little more then the title to work with.  Your answer about remote calls into the API functions is not what I'm looking for.
Is my request clear enough ?

Just to be clear, what I'm requesting is to have user-defined parameters/variables that I can use in the node firmware and to be able to change the value of these parameters remotely "OTA".  Idealy these parameters are persisted into the eeprom memory after they changed and are extracted from the eeprom memory at startup.

Let me know if you need more information.

user2684, 2017-07-15:
Ok, this may be something differently actually. The remote api allows you to call whatever function available to configure a sensor remotely by providing parameters. You want instead a way to save a parameter in EEPROM and loaded it at startup. Would my mind sharing an example just for me to understand better the use case? Thanks!

stefaanv, 2017-07-16:
Below an example of some work in progress.  The interesting part is the receive() function : I'm parsing the V_TEXT (child ID =200) message there and extracting the named parameter.
I'm not yet storing the parameter value in eeprom, but that would be very helpfull.
Nice thing is that the parameter can be set/read through OH2 by using the MQTT eventbus interface.

```
// Enable and select radio type attached
#define MY_RADIO_NRF24
#define NODE_ID 4

#include <MySensors.h>
#include <TimerOne.h>
#define SENSOR_PIN 3 
#define LED_PIN        A2
#define CHILD_ID_GAS 1              // Id of the sensor child
#define CHILD_ID_TEXT 200              // Id of the sensor child

unsigned long debounceTimerPeriod = 20000;	// 20ms : Debounce polling periode [µs]
unsigned long MaxInterPulseTime = 600;	// [s] : Max time between pulses

volatile unsigned long pulseCounter = 0;
volatile unsigned long lastRaisingEdge = 0;
volatile unsigned long lastFallingEdge = 0;
volatile bool pinStatus = false;
volatile byte shiftRegister = 0;
volatile unsigned long timeBetweenPulses = 0;
volatile unsigned long pulseLength = 0;
unsigned long lastPulseCounter = 0;
bool zeroConsumption = false;

MyMessage counterMsg(CHILD_ID_GAS, V_VAR1);
MyMessage timeBetweenMsg(CHILD_ID_GAS, V_VAR2);
MyMessage pulseLengthMsg(CHILD_ID_GAS, V_VAR3);
MyMessage textMsg(CHILD_ID_TEXT, V_TEXT);

void presentation()
{
	present(CHILD_ID_GAS, S_CUSTOM);
	present(CHILD_ID_TEXT, S_INFO);
	request(CHILD_ID_GAS, V_VAR1);
}

void setup()
{
	Serial.begin(115200);
	Serial.println("Starting ...");
	pinMode(LED_PIN, OUTPUT);
	pinMode(SENSOR_PIN, INPUT_PULLUP);
	digitalWrite(LED_PIN, HIGH);
	wait(500);
	digitalWrite(LED_PIN, LOW);
	if (digitalRead(SENSOR_PIN) == LOW)
	{
		wait(500);
		digitalWrite(LED_PIN, HIGH);
		wait(500);
		digitalWrite(LED_PIN, LOW);
	}
	Timer1.initialize(debounceTimerPeriod); 
	Timer1.attachInterrupt(onTick);
	Serial.println("Setup done");
	send(textMsg.set("Setup done"));

	Timer1.start();
}

void loop()
{
	if (pulseCounter < 1000) {
		request(CHILD_ID_GAS, V_VAR1);
		send(textMsg.set("Requesting pulseCounter value"));
	}

	if (pulseCounter>0 && lastPulseCounter==0)
	{
		zeroConsumption = true;
		lastPulseCounter = pulseCounter;
		Serial.println();
		Serial.print("received new PC : ");
		Serial.print(pulseCounter);
		Serial.print(", Zero consumption period started");
		Serial.println();
	}

	if (lastPulseCounter != pulseCounter) {
		digitalWrite(LED_PIN, HIGH);
		send(counterMsg.set(pulseCounter));
		Serial.println();
		Serial.print("PC:");
		Serial.print(pulseCounter);

		send(pulseLengthMsg.set(zeroConsumption?0:pulseLength));
		Serial.print(", PL:");
		Serial.print(zeroConsumption ? 0 : pulseLength);
		send(timeBetweenMsg.set(zeroConsumption?0:timeBetweenPulses));
		Serial.print(", TBP:");
		Serial.print(zeroConsumption ? 0 : timeBetweenPulses);
		lastPulseCounter = pulseCounter;
		zeroConsumption = false;
		Serial.println(", Zero consumption period ended");

		digitalWrite(LED_PIN, LOW);
	}

	Serial.print(".");

	if (!zeroConsumption && ((millis() - lastFallingEdge)/1000 > MaxInterPulseTime)) {
		digitalWrite(LED_PIN, HIGH);
		zeroConsumption = true;
		send(counterMsg.set(pulseCounter));
		send(pulseLengthMsg.set(0));
		send(timeBetweenMsg.set(0));
		Serial.println();
		Serial.print("PC:");
		Serial.print(pulseCounter);
		Serial.print(", Zero consumption period started");
		Serial.println();

		lastPulseCounter = pulseCounter;
		digitalWrite(LED_PIN, LOW);
	}

	wait(1000);
}

void receive(const MyMessage &message)
{
	Serial.print(F("Received: sensor="));
	Serial.print(message.sensor);
	Serial.print(F(", type="));
	Serial.print(message.type);
	Serial.print(F(", payload='"));
	Serial.print(message.data);
	Serial.println(F("'"));
	if (message.type == V_VAR1)
	{
		pulseCounter = message.getLong();
		Serial.print("Setting pulseCounter to ");
		Serial.println(pulseCounter);
		send(textMsg.set("Received pulseCounter value"));
	}

	if (message.type == V_TEXT)
	{
		String all = message.data;
		int equalSignPosition = all.indexOf('=');
		if (equalSignPosition < 0) return;
		String key = all.substring(0, equalSignPosition);
		String value = all.substring(equalSignPosition + 1, all.length());

		if (key == "debounceTimerPeriod")
		{
			int dbPvalue = value.toInt();
			if (dbPvalue > 5 && dbPvalue < 5000) {
				Serial.print("Setting the debounce timer period to ");
				Serial.print(dbPvalue);
				Serial.println("ms");
				debounceTimerPeriod = dbPvalue;
				Timer1.setPeriod(debounceTimerPeriod);
				SendParameter("Debounce period", debounceTimerPeriod);
			}
		}
		if (key == "info")
		{
			Serial.print("Sending information");
			SendParameter("debouncePeriod", debounceTimerPeriod);
			SendParameter("pulseCounter", pulseCounter);
		}
	}
}

void SendParameter(char* parameterName, unsigned long variable)
{
	char buffer[30];
	sprintf(buffer, "%s->%lu", parameterName, variable);
	Serial.println(buffer);
	send(textMsg.set(buffer));
}

void onTick()
{
	unsigned long now = millis();
	pinStatus = (digitalRead(SENSOR_PIN) == LOW);
	shiftRegister = (shiftRegister >> 1) | (pinStatus ? 0x80 : 0x00) ;
	if (shiftRegister == 0x80)	//raising edge
	{
		timeBetweenPulses = now - lastRaisingEdge;
		lastRaisingEdge = now;
		pulseCounter++;
	}
	if (shiftRegister == 0x7F)	//falling edge
	{
		pulseLength = now - lastRaisingEdge;
		lastFallingEdge = now;
	}
}

```

user2684, 2017-07-22:
Hi, from what I see it really looks like the remote api already part of the development branch (https://github.com/mysensors/NodeManager/tree/development#communicate-with-nodemanager-and-its-sensors). For almost every function mapped, when a V_CUSTOM message is received, this is parsed and the payload is used to invoke the function so doing something similar to what I see above if I'm not completely mistaken. I'm just not using strings since would have an impact on the storage utilization but integers for mapping commands to functions. Am I wrong? Thanks!

stefaanv, 2017-07-23:
@user2684 
OK for calling API functions, but does it also provide a way to change program variables ?  I need to set (my own) program variables.

user2684, 2017-07-23:
This is what the remote API does under the hood. The message gets in through NodeManager::received() -> Sensor::process() -> Sensor::onProcess() and eventually set a variable local to the instance. So as far as the variable is local to the a built-in sensor, this should do the job. If we are talking about a custom sensor, this would require to be registered and then you can implement your own onProcess(). If we are talking about an external variable this can be probably done with a similar approach but probably would be easier to just implement your own code in the main receive() outside NodeManager. Let me know of course if I have misunderstood anything :) Thanks!

user2684, 2018-03-31:
Closing this for now. If you think the remote API is not covering this request, feel free to reopen it. Thanks!

closed 2018-03-31 by user2684