mysensors/NodeManager #170: DHT sensor is not working properly

user2684, 2017-07-15:
With both 1.5 and 1.6-dev
```
MY I=3 M=1
DHT I=2 T=nan
SLEEP 60s
```

user2684, 2017-07-16:
Tested with DHT22 and it is working fine both temperature and humidity

SergioRius, 2017-07-17:
Entire failing folder attached
[NodeManager_MultiSensor_v1.0.zip](https://github.com/mysensors/NodeManager/files/1152731/NodeManager_MultiSensor_v1.0.zip)

user2684, 2017-07-22:
I'm experiencing a different issue but it may be related with #177. Investigation in progress

user2684, 2017-07-29:
This should be fixed by #175. Looks like the MySensors's DHT library as you were suggesting works way better and fixes many issues. If the problem persists, feel free to re-open this issue
Thanks

closed 2017-07-29 by user2684

getlarge, 2017-08-02:
Could we keep the Adafruit DHT library for DHT21 and AM2305 (and maybe others) ?
The MySensors one does not work with AM2305...

user2684, 2017-08-02:
DHT21 got lost while migrating to MySensors library, my mistake, sorry! I've added #196, the idea is to add a DHT21 specific module so to load the Adafruit's library just for this. Not great but I cannot think of a better compromise since a standard DHT sensor with the Adafruit's library has a lot of issues. Thanks for pointing this out!

PR #175 (2017-07-29): DHT Update
* Use the inbuilt DHT library conversion facility (fixes #148)
* Fixes #144 
* Fixes #145 
* Fixes #176 
* Fixes related issue of #177
* Fixes #170