mysensors/NodeManager #178: Sending a message directly to another node

rakeshpai, 2017-07-18:
I haven't thought this through - I'm just posting here for discussion at this time. This might be unfeasable, or outside the scope of NodeManager, but I thought it's worth bringing up.

The scenario: The most popular style of using MySensors is that nodes send all 'events' to the gateway, the gateway forwards those messages to the controller, the controller decides what action to take, and then messages flow back to the concerned nodes. A drawback with this setup is that the controller is a required part of the setup, and everything stops working if the controller is unavailable. The controller is also probably the most complex part of the running system - it needs a full computer with all it's services/daemons running, and a ethernet/wifi network stack - and is more likely to fail than the MySensors nodes.

Instead, we could write our sketches with some simple rules baked in. For example, if a motion sensor detects motion, and uses its LDR to check that the room is dark, it could send a message directly to a lighting node to turn on the lights, without having to involve a controller.

You could think of it as distributing the most important controller actions to the nodes. The controller still has it's place - it's still logging all activity, and can take some of the more complex decisions - but the 'instinctual' actions can be fired off directly from the nodes.

While the exact logic will be different for different use-cases and will be left up to the sketch implementer, we'd need the ability to read sensor data and message other nodes directly for this to work.

There are things I still haven't thought about. (For example, if the logic changes, or if the list of message destinations changes, the device has to be re-flashed, which isn't great.) But I'm not the expert here ;) so I thought I'd make the suggestion and see if this is something that appeals to you.

user2684, 2017-07-22:
Hi, it is definitely a good idea but from what I understand it is dependent on the transport layer of the core library. I think every message is routed through a gateway or repeater and I remember there are a few threads (e.g. https://forum.mysensors.org/topic/826/node-to-node-communication) on the forum articulating around the technical details and why it has been implemented in this way. I guess the issue is the complexity of having a fully mashed network but probably somebody from the core development team can be more accurate :-)

rakeshpai, 2017-07-22:
Depending on the gateway or on repeaters for message routing is fine, IMHO. They are points of failure, sure, but once installed correctly they should work trouble-free, theoretically at least. Computers (where the controller would run) are far more troublesome to keep running 24x7. It's never as simple as run-and-forget, and requires some amount of regular maintenance and monitoring. That's fine for the more complex automation (like 'switch my living room to movie watching mode'), but for the simplest operations (eg. 'turn on lights when motion is detected'), we can improve reliability by not taking a hard dependency on the controller, and instead doing some decision-making locally on the node.

I don't mean to overstate the problem - it isn't too hard to keep a computer running after all. But if there's an opportunity to reduce dependencies and improve reliability, why not take it. ;)

stefaanv, 2017-07-23:
NOT a good idea IMHO.  
The trend in embedded development is to move away from this kind of architectures.  The reason being that embedded software is harder to make, debug and change.  You want to make embedded software as simple as possible and put the more complex logic in system where you have all the development and management luxuries.
KNX started out as a distributed logic system but is now mainly used as a communications protocol.
My advise : don't go that way.

dudman8, 2018-02-25:
I made a few changes/enhancements or possible just complicated things :) to some origional code on mysensors forum. Heres a work in progress if its interesting, basicly I put it into a seperate library, and added a few features https://github.com/dudman8/MysensorsPairNodeLocal

user2684, 2018-03-31:
Closing this since implemented through #312. Feel free to reopen it if you think something is missing. Thanks!

closed 2018-03-31 by user2684