mysensors/NodeManager #182: Remote Config not working if node uses Sleep Mode

kruemelro, 2017-07-23:
Hi,

i flash some Nodes with Nodemanager and wan't them to configure with the remote config-messages. As long as the Nodes are in idle Mode they response to the config messages as supposed. 
If i set them to Mode1 they sleep for the configured time but doesn't response to config Messages anymore. 
I use a MQTT-Gateway with MyController. In MyController the Node is marked as a SmartSleep Node, so MyController waits with Messages to send until the Node comes out of sleep and then send the Message.
The Log of the Gateway show the Config Message is sent to the Node:

0;255;3;0;9;TSF:MSG:READ,8-8-0,s=1,c=1,t=0,pt=7,l=5,sg=0:26.90
0;255;3;0;9;Sending message on topic: mygateway-out/8/1/1/0/0
0;255;3;0;9;TSF:MSG:READ,8-8-0,s=2,c=1,t=1,pt=7,l=5,sg=0:58.87
0;255;3;0;9;Sending message on topic: mygateway-out/8/2/1/0/1
0;255;3;0;9;TSF:MSG:READ,8-8-0,s=3,c=1,t=4,pt=7,l=5,sg=0:956.88
0;255;3;0;9;Sending message on topic: mygateway-out/8/3/1/0/4
0;255;3;0;9;TSF:MSG:READ,8-8-0,s=4,c=1,t=5,pt=0,l=6,sg=0:stable
0;255;3;0;9;Sending message on topic: mygateway-out/8/4/1/0/5
0;255;3;0;9;TSF:MSG:READ,8-8-0,s=255,c=3,t=22,pt=5,l=4,sg=0:74711
0;255;3;0;9;Sending message on topic: mygateway-out/8/255/3/0/22
0;255;3;0;9;Message arrived on topic: mygateway-in/8/200/2/0/48
0;255;3;0;9;!TSF:MSG:SEND,0-0-8-8,s=200,c=2,t=48,pt=0,l=7,sg=0,ft=0,st=NACK:VERSION
0;255;3;0;9;Message arrived on topic: mygateway-in/8/255/3/0/18
0;255;3;0;9;!TSF:MSG:SEND,0-0-8-8,s=255,c=3,t=18,pt=0,l=0,sg=0,ft=0,st=NACK:

but the Node never Response to them.

user2684, 2017-07-23:
Hi, thanks for reporting this issue. May you please collect and share NodeManager's logs from the serial output when facing this scenario? I would like to see if the node actually receives this message and should this happen we should see a RECV log entry. Thanks!

kruemelro, 2017-07-23:
Hi, 
thanks, i haven't had the serial-debug on my mind. I switched it on and, as you mentioned, the Node don't receive the Message. 
I know the smartsleep function from mySensors waits 500ms befor go to sleep, but the SLEEP message from NodeManager cames instant afer the sensor loop. I inserted a wait code into the sleep function at NodeManager.cpp and know it works as it should.

Here a snippet from what i've changed:

```
//wrapper of smart sleep
void NodeManager::_sleep() {
  // reset the last interrupt pin
  _last_interrupt_pin = -1;
  // wait for incomming config Messages
  #if REMOTE_CONFIGURATION == 1
      #if DEBUG == 1
          Serial.println(F("WAIT FOR CONFIG"));
      #endif
      wait(500);
  #endif 
  // calculate the seconds to sleep
  long sleep_sec = _sleep_time;
```

user2684, 2017-07-23:
Thanks for the feedback, strange  because NodeManager uses smart sleep only. I had a similar situation when the node was a bit too far away from the gw and when waking up was re-tramsmitting the message multiple times hence ignoring received message. If you are in a similar situation, I'd try moving the node closer to the gw and alternatively increasing MY_SMART_SLEEP_WAIT_DURATION_MS. Just because I'd try to avoid the workaround if possible otherwise you'll face the same at the next upgrade. Thanks!

kruemelro, 2017-07-28:
Hi,

i get your point. i removed the Code and set the MY_SMART_SLEEP_WAIT_DURATION_MS to 800ms. Now it is working :-). Thanks

user2684, 2017-07-29:
Great! I'd recommend anyway to enable MySensors debug on the node and check if there are re-transmission because if so the issue could show up again :)

closed 2017-07-29 by user2684