mysensors/NodeManager #186: Allow enabling internal pullup resistors

rakeshpai, 2017-07-29:
For simple switch-like sensors, like door sensors, switches, etc., we can enable the internal pullups in the microcontroller to avoid the use of external pullup resistors. It's easy to do (`pinMode(INPUT_PULLUP)`), and reduces component count.

[Example](https://github.com/mysensors/MySensorsArduinoExamples/blob/master/examples/BinarySwitchSensor/BinarySwitchSensor.ino)

user2684, 2017-07-29:
Hi, there should be a setInitial() or equivalent function when having a pull up applies if I'm not wrong. Even if I'm not using INPUT_PULLUP but just INPUT followed by a DigitalWrite() it should be equivalent I guess. Am I wrong?
Thanks!

rakeshpai, 2017-07-29:
Sorry, I didn't realise that setInitial was configuring the pullup resistors. PEBCAK - you have documented it correctly too. You might want to consider changing the name - I didn't associate the name with pullups. Closing this anyway - it certainly does what I need. Thanks!

closed 2017-07-29 by rakeshpai