mysensors/NodeManager #196: Add support for DHT21

user2684, 2017-08-02:
support got in and out with #175. Use Adafruit's library for this

getlarge, 2017-08-03:
Well, it seems i got worried for nothing! I used DHT21 in Adafruit library to read the AM2305. But in fact it gets better reading with this library. so let's see with the real DHT21 (AM2301).

If it didn't, i thought of the same thing : create another module!

user2684, 2017-08-04:
Oh cool, thanks, can I close this down then?

getlarge, 2017-08-04:
Yes, but i would be probably be better to verify with a DHT21 (AM2301) owner that this lib works for them too ?

user2684, 2017-08-05:
Good point. I'll keep this open until this will be tested (unfortunately I do not have such a sensor) but I'll change the milestone to v1.7. Thanks!

closed 2018-02-11 by user2684