mysensors/NodeManager #202: New Chirp Sensor

betonmoewe, 2017-08-16:
Hello,

Here are 2 diffs and my .ino as an example for including the (soil moisture) Chirp sensor into the NodeManager (1.6):

[Chirp_Sensor.zip](https://github.com/mysensors/NodeManager/files/1228472/Chirp_Sensor.zip)

The solution can only handle 1 chirp sensor per node!

If you want, fell free to put this into version 1.7 ...

More informations at: https://www.tindie.com/products/miceuz/i2c-soil-moisture-sensor/

and the library I use is: https://github.com/Apollon77/I2CSoilMoistureSensor

I use this kind of sensor since ca. 1 year to monitor the soil of my plants.

If you need more explanation...

Best regards

Betonmoewe

user2684, 2017-08-16:
Thanks for the contribution! I've added this to v1.7 backlog

bilbolodz, 2017-08-18:
Working (for example):
`
[..]
#include <I2CSoilMoistureSensor.h>
#include <Wire.h>
#include <MySensors.h>

#define DEFAULT_min_hum_adc 345
#define DEFAULT_max_hum_adc 560

void presentation()
{
[..]
present(CHILD_ID_HUM, S_HUM, "Humidity");
present(CHILD_ID_TEMP, S_TEMP, "Tempature");
}

void loop()
{
[..]
//When time comes read sensor...
  while (sensor.isBusy()) wait(50);
  float tempC = sensor.getTemperature() / (float)10;

 int sensorValueM;
  //Firs reading gives old value so we need read twice!
  while (sensor.isBusy()) wait(50);
  sensorValueM = sensor.getCapacitance();

  while (sensor.isBusy()) wait(50);
  sensorValueM = sensor.getCapacitance();

  sensor.sleep();

/*
DEFAULT_min_hum_adc and DEFAULT_max_hum_and  should be calibrated for particular sensor AND THESE numbers change in case of power supply voltage change. In case of battery operated device there should be compensation algorithm used (not ready yet in my device - work in progress) 
*/

  int moistureLevel = min(map(sensorValueM, DEFAULT_min_hum_adc, DEFAULT_max_hum_adc, 0, 100), 100);

  moistureLevel = max(0, moistureLevel);
[..]
send(msgTemp.set(tempC, 1));
 send(msgHum.set(moistureLevel));
[..]
}
`

user2684, 2017-08-20:
Thanks, this will be definitely useful!

betonmoewe, 2017-08-20:
Hi,
 that‘s nearly the way, I implemented the CHIRP class! But thanks, now I have integrated the 2times reading for the humidity too.
@user2684: I will upload the new diffs soon. I have tried to make my own branch at Github, but I failed completely ...

user2684, 2017-08-21:
If you need any help with Github, I've tried to put together some guidance here: https://github.com/mysensors/NodeManager/tree/development#contributing
I'm a newbie as well so that one is not necessarily the best way but at least worked for me :)

PR #283 (2018-02-11): Added support for CHIRP soil moisture sensor
Fixes #202
closed 2018-02-11 by user2684