mysensors/NodeManager #205: no loop after sleep any more

betonmoewe, 2017-08-17:
hi,

very very strange problem ... I was playing with some additions /changes for the chirp sensor. Yesterday everything was fine but since today, I have a BIG problem (even with the code from yesterday!!!). The sensor starts up normally with his first readings. After this, the sensor goes to sleep and after the sleep time, it wakes up and goes to sleep again directly after the wakeup without doing the loop uf the sensor (even no simple println). I have reapplied my code from scatch again to make shure, that there is no other error ....
No way to get the sensor working again ... (my old mysensor sketch wich was installed as test works still like a sharm)

I have really no idea 

Betonmoewe

user2684, 2017-08-18:
Really strange. Do you have any logs to share? Thanks

betonmoewe, 2017-08-18:
hi 

here are some logs (sleep time 60 sec. but the same behaviour with longer sleep time)

`
0 MCO:BGN:INIT NODE,CP=RNNNA--,VER=2.1.1
40 MCO:BGN:BFR
REG I=1 P=16 P=7 T=1

REG I=2 P=16 P=6 T=0

REG I=3 P=16 P=16 T=23

NodeManager v1.6-dev

LIB V=2.1.1 R=N T=N A=A S=- B=-

INT P=3 M=255

INT P=2 M=255

149 TSM:INIT
231 TSF:WUR:MS=0
253 TSM:INIT:TSP OK
274 TSF:SID:OK,ID=103
299 TSM:FPAR
348 TSF:MSG:SEND,103-103-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
927 TSF:MSG:READ,0-0-103,s=255,c=3,t=8,pt=1,l=1,sg=0:0
985 TSF:MSG:FPAR OK,ID=0,D=1
1286 TSF:MSG:READ,110-110-103,s=255,c=3,t=8,pt=1,l=1,sg=0:1
2424 TSM:FPAR:OK
2441 TSM:ID
2455 TSM:ID:OK
2469 TSM:UPL
2490 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=OK:1
2566 TSF:MSG:READ,0-0-103,s=255,c=3,t=25,pt=1,l=1,sg=0:1
2625 TSF:MSG:PONG RECV,HP=1
2656 TSM:UPL:OK
2672 TSM:READY:ID=103,PAR=0,DIS=1
2715 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
2795 TSF:MSG:READ,0-0-103,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
2871 TSF:MSG:SEND,103-103-0-0,s=255,c=0,t=17,pt=0,l=5,sg=0,ft=0,st=OK:2.1.1
2953 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=6,pt=1,l=1,sg=0,ft=0,st=OK:0
3026 TSF:MSG:READ,0-0-103,s=255,c=3,t=6,pt=0,l=1,sg=0:M
RADIO OK

3090 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=11,pt=0,l=14,sg=0,ft=0,st=OK:ChirpSensor_NM
3192 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=12,pt=0,l=3,sg=0,ft=0,st=OK:1.0
PRES I=200, T=23

3270 TSF:MSG:SEND,103-103-0-0,s=200,c=0,t=23,pt=0,l=0,sg=0,ft=0,st=OK:
PRES I=201, T=30

3364 TSF:MSG:SEND,103-103-0-0,s=201,c=0,t=30,pt=0,l=0,sg=0,ft=0,st=OK:
BATT V=3.41 P=100

SEND D=0 I=201 C=0 T=38 S= I=0 F=3.41

3526 !MCO:SND:NODE NOT REG
3616 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=0,pt=1,l=1,sg=0,ft=0,st=OK:100
PRES I=1 T=7

3694 TSF:MSG:SEND,103-103-0-0,s=1,c=0,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
PRES I=2 T=6

3780 TSF:MSG:SEND,103-103-0-0,s=2,c=0,t=6,pt=0,l=0,sg=0,ft=0,st=OK:
PRES I=3 T=16

3868 TSF:MSG:SEND,103-103-0-0,s=3,c=0,t=16,pt=0,l=0,sg=0,ft=0,st=OK:
READY

3950 MCO:REG:REQ
3981 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=26,pt=1,l=1,sg=0,ft=0,st=OK:2
4057 TSF:MSG:READ,0-0-103,s=255,c=3,t=27,pt=1,l=1,sg=0:1
4116 MCO:PIM:NODE REG=1
4143 MCO:BGN:STP
MY I=103 M=1

I2C Soil Moisture Sensor Address: 20

Sensor Firmware version: 23

5185 MCO:BGN:INIT OK,TSP=1
CHIRP I=1 H=0.00 H_orig=329.00 Offset=330 Range=520 Normalized=1

SEND D=0 I=1 C=0 T=1 S= I=0 F=0.00

5373 TSF:MSG:SEND,103-103-0-0,s=1,c=1,t=1,pt=7,l=5,sg=0,ft=0,st=OK:0.00
CHIRP I=2 T=24.30

SEND D=0 I=2 C=0 T=0 S= I=0 F=24.30

5550 TSF:MSG:SEND,103-103-0-0,s=2,c=1,t=0,pt=7,l=5,sg=0,ft=0,st=OK:24.30
CHIRP I=3 L=20107.00

SEND D=0 I=3 C=0 T=23 S= I=0 F=20107.00

8730 TSF:MSG:SEND,103-103-0-0,s=3,c=1,t=23,pt=7,l=5,sg=0,ft=0,st=OK:20107.00
SLEEP 60s

8867 MCO:SLP:MS=60000,SMS=1,I1=255,M1=255,I2=255,M2=255
8941 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=22,pt=5,l=4,sg=0,ft=0,st=OK:4796
9519 MCO:SLP:TPD
9535 MCO:SLP:WUP=-1
AWAKE

SLEEP 60s

9558 MCO:SLP:MS=60000,SMS=1,I1=255,M1=255,I2=255,M2=255
9641 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=22,pt=5,l=4,sg=0,ft=0,st=OK:5492
10219 MCO:SLP:TPD
10237 MCO:SLP:WUP=-1
AWAKE

SLEEP 60s

10260 MCO:SLP:MS=60000,SMS=1,I1=255,M1=255,I2=255,M2=255
10342 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=22,pt=5,l=4,sg=0,ft=0,st=OK:6197
10924 MCO:SLP:TPD
10942 MCO:SLP:WUP=-1
AWAKE

SLEEP 60s

10964 MCO:SLP:MS=60000,SMS=1,I1=255,M1=255,I2=255,M2=255
11046 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=22,pt=5,l=4,sg=0,ft=0,st=OK:6901
11626 MCO:SLP:TPD
11644 MCO:SLP:WUP=-1
AWAKE

SLEEP 60s

11667 MCO:SLP:MS=60000,SMS=1,I1=255,M1=255,I2=255,M2=255
11749 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=22,pt=5,l=4,sg=0,ft=0,st=OK:7604
12331 MCO:SLP:TPD
12349 MCO:SLP:WUP=-1
AWAKE

SLEEP 60s

12371 MCO:SLP:MS=60000,SMS=1,I1=255,M1=255,I2=255,M2=255
12460 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=22,pt=5,l=4,sg=0,ft=0,st=OK:8308
13039 MCO:SLP:TPD
13058 MCO:SLP:WUP=-1
AWAKE

SLEEP 60s

13080 MCO:SLP:MS=60000,SMS=1,I1=255,M1=255,I2=255,M2=255
13176 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=22,pt=5,l=4,sg=0,ft=0,st=OK:9017
13756 MCO:SLP:TPD
13774 MCO:SLP:WUP=-1
AWAKE

SLEEP 60s

13797 MCO:SLP:MS=60000,SMS=1,I1=255,M1=255,I2=255,M2=255
13883 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=22,pt=5,l=4,sg=0,ft=0,st=OK:9734
14465 MCO:SLP:TPD
14483 MCO:SLP:WUP=-1
AWAKE

SLEEP 60s

14505 MCO:SLP:MS=60000,SMS=1,I1=255,M1=255,I2=255,M2=255
14587 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=22,pt=5,l=4,sg=0,ft=0,st=OK:10442
15167 MCO:SLP:TPD
15185 MCO:SLP:WUP=-1
AWAKE

SLEEP 60s

15208 MCO:SLP:MS=60000,SMS=1,I1=255,M1=255,I2=255,M2=255
15290 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=22,pt=5,l=4,sg=0,ft=0,st=OK:11145
15872 MCO:SLP:TPD
15890 MCO:SLP:WUP=-1
AWAKE

SLEEP 60s

15912 MCO:SLP:MS=60000,SMS=1,I1=255,M1=255,I2=255,M2=255
15994 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=22,pt=5,l=4,sg=0,ft=0,st=OK:11849
16574 MCO:SLP:TPD
16592 MCO:SLP:WUP=-1
AWAKE

SLEEP 60s

16615 MCO:SLP:MS=60000,SMS=1,I1=255,M1=255,I2=255,M2=255
16699 TSF:MSG:SEND,103-103-0-0,s=255,c=3,t=22,pt=5,l=4,sg=0,ft=0,st=OK:12552
17281 MCO:SLP:TPD
`
[ChirpSensor_NM.zip](https://github.com/mysensors/NodeManager/files/1234942/ChirpSensor_NM.zip)

and the related source ...

Best regards

Betonmoewe (Frank)

betonmoewe, 2017-08-18:
by the way, I just tested it just with one Class instance (only Humidity)... same result :(

betonmoewe, 2017-08-19:
ok, I have done some debugging and after a while, I got a hint: I have set a sleeptime (setSleepMinutes) of 1 minute but I have set a reporttime (setReportInterval) of 60 minutes. NoneManager ignores the sleeptime for reporting the sensor readings, it depends only on the reporttime: i.e. setting the report time to 1 minute too, everything is fine.

ok my misinterpretation of this two timers was, that after sleeptime is over, the sensor reports its data too ... I now don't understand why I should set a sleep time different than reporting time

Best regards

Betonmoewe

closed 2017-08-19 by betonmoewe

user2684, 2017-08-20:
Sorry once again for the late reply. Sleep and reporting intervals have been decoupled in v1.6 (I must admit this has not been made very clear so far). Since for non-sleeping boards the behavior was different, I tried to make it coherent regardless of the sleep configuration. So reporting time (by default 10 minutes) applies in any case and if the board is sleeping does not have a direct impact on the time the board will report back a new measure. It is somehow documented here https://github.com/mysensors/NodeManager/tree/development#set-reporting-intervals-and-sleeping-cycles but I'll try to make it more clear. Thanks!