mysensors/NodeManager #209: build for esp8266 not possible

enny1170, 2017-08-23:
if i try to build nodemanager for esp based boards with ESP librarys V2.3.0. running in fallowing Errors:

sketch\NodeManager.cpp: In function 'float getVcc()':
NodeManager.cpp:22: error: 'ADMUX' was not declared in this scope
       ADMUX = (_BV(REFS0) | _BV(MUX3) | _BV(MUX2) | _BV(MUX1));
       ^

In file included from sketch\NodeManager.h:7:0,
                 from sketch\NodeManager.cpp:5:
NodeManager.cpp:22: error: 'REFS0' was not declared in this scope
       ADMUX = (_BV(REFS0) | _BV(MUX3) | _BV(MUX2) | _BV(MUX1));
                    ^
\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:184:25: note: in definition of macro '_BV'
 #define _BV(b) (1UL << (b))
                         ^
NodeManager.cpp:22: error: 'MUX3' was not declared in this scope
       ADMUX = (_BV(REFS0) | _BV(MUX3) | _BV(MUX2) | _BV(MUX1));
                                 ^

\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:184:25: note: in definition of macro '_BV'
 #define _BV(b) (1UL << (b))
                         ^
NodeManager.cpp:22: error: 'MUX2' was not declared in this scope
       ADMUX = (_BV(REFS0) | _BV(MUX3) | _BV(MUX2) | _BV(MUX1));
                                             ^
Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:184:25: note: in definition of macro '_BV'
 #define _BV(b) (1UL << (b))
                         ^
NodeManager.cpp:22: error: 'MUX1' was not declared in this scope
       ADMUX = (_BV(REFS0) | _BV(MUX3) | _BV(MUX2) | _BV(MUX1));
                                                         ^
\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:184:25: note: in definition of macro '_BV'
 #define _BV(b) (1UL << (b))
                         ^
NodeManager.cpp:27: error: 'ADCSRA' was not declared in this scope
     ADCSRA |= _BV(ADSC);
     ^

\NodeManager.h:7:0,
                 from sketch\NodeManager.cpp:5:
NodeManager.cpp:27: error: 'ADSC' was not declared in this scope
     ADCSRA |= _BV(ADSC);
                   ^
\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:184:25: note: in definition of macro '_BV'
 #define _BV(b) (1UL << (b))
                         ^

NodeManager.cpp:28: error: 'bit_is_set' was not declared in this scope
     while (bit_is_set(ADCSRA, ADSC)) {};
                                   ^
NodeManager.cpp:30: error: 'ADC' was not declared in this scope
     return (float)((1125300UL) / ADC) / 1000;
                                  ^
sketch\NodeManager.cpp: In member function 'int NodeManager::registerSensor(Sensor*)':
NodeManager.cpp:2096: error: return-statement with no value, in function returning 'int' [-fpermissive]
   if (sensor->getChildId() > MAX_SENSORS) return;
                                           ^

\sketch\NodeManager.cpp: In member function 'bool NodeManager::renameSensor(int, int)':
NodeManager.cpp:2136: error: return-statement with no value, in function returning 'bool' [-fpermissive]
   if (old_child_id > MAX_SENSORS || new_child_id > MAX_SENSORS) return;
                                                                 ^
\sketch\NodeManager.cpp: In member function 'void NodeManager::before()':
NodeManager.cpp:2197: error: 'INTERNAL' was not declared in this scope
     if (! _battery_internal_vcc && _battery_pin > -1) analogReference(INTERNAL);
                                                                      ^
\sketch\NodeManager.cpp: In member function 'void NodeManager::_process(const char*)':
NodeManager.cpp:2410: error: 'WDTCSR' was not declared in this scope
       WDTCSR |= (1<<WDCE) | (1<<WDE);
       ^
NodeManager.cpp:2410: error: 'WDCE' was not declared in this scope
       WDTCSR |= (1<<WDCE) | (1<<WDE);
                     ^
NodeManager.cpp:2410: error: 'WDE' was not declared in this scope
       WDTCSR |= (1<<WDCE) | (1<<WDE);

enny1170, 2017-08-23:
Sorry ist solved, some missconfig in config.h

closed 2017-08-23 by enny1170

user2684, 2017-08-23:
No problem, thanks for looking at it anyway :)

clempat, 2017-10-30:
@enny1170 May I ask you what was the missconfiguration ? I am getting the same errors.