mysensors/NodeManager #210: Optimize memory utilization

user2684, 2017-08-26:
* https://forum.mysensors.org/topic/7266/nodemanager-v1-6-dev-comments-and-issues/11
* https://forum.mysensors.org/topic/7212/integrating-nodemanager-with-sketch-generator/25

Akubi, 2017-09-17:
I’m facing memory problems with one node. I have 6 child and I’m out of memory.
In fact I have collision between Stack and Heap. Free memory is too low to work correctly.

My estimation is that 1 Sensor child use ~200bytes of memory
-	70 for member variable
-	66 for 2 MyMessage
-	60 for virtual method

So 6 Child = more than 1Kb.
I’m working on a pull request to remove the 2 MyMessages. 
So in my configuration I will spare ~400bytes of RAM.

Can I proceed with that or are you changing software structure working on Issue #210 ?

user2684, 2017-09-24:
Hi, if referring to #198, I guess it may change something around the data structures used so I'd rather hold on to avoid you spend time on something which may change. But I'd love to see these changes applied just after , memory utilization is the first issue we have with Nodemanager right now honestly and any suggestion is more than welcome!

Akubi, 2017-09-24:
Hi, yes and no.

In fact, in order to decrease the amount of used memory in a node with multiple child I tested a modified version of NodeManager.

The idea is to have only one message structure for outgoing messages in the system. In order to achieve that I changed some structures and some methods.

Mainly :

In sensor Class, I remove the _msg_service message end changed the _msg to a message pointer.

```
/***************************************
   Sensor: generic sensor class
*/
class Sensor {
…
  protected:
    MyMessage* _msg;
    //MyMessage _msg_service;
```

I then splitt the send function :
```
void _send(MyMessage & msg);
void _sendService(MyMessage & msg);
```

In NodeManager Class, I add 2 methods in order to use NodeManager capability to send messages from sensor Class. 

```
/***************************************
   NodeManager: manages all the aspects of the node
*/
class NodeManager {
  public:
…
	MyMessage* getMessage();
	void sendMessage();
```

In Sensor Class :
Constructor I remove messages instantiation and I add :

`  _msg = _node_manager->getMessage();`

Send methodes :

```
// send a message to the network
void Sensor::_send(MyMessage & message) {
	// setup the message
	message.setSensor(_child_id);
	message.setType(_type);
	_node_manager->sendMessage();
}

// send a message to the network
void Sensor::_sendService(MyMessage & message) {
	// setup the message
	message.setSensor(_child_id);
	message.setType(V_CUSTOM);
	_node_manager->sendMessage();
}
```

Small change in setType method :

```
void Sensor::setType(int value) {
  _type = value;
  //_msg.setType(_type);
}
```

And then in order to send messages :

`_send(_msg->set(avg, _float_precision));`
Or service message
`_sendService(_msg->set(function));`

Of course the send implementation in NodeManager  is :

```
// send a message to the network
void NodeManager::_send(MyMessage & message) {
	// setup the message
	message.setSensor(CONFIGURATION_CHILD_ID);
	message.setType(V_CUSTOM);
	sendMessage();
}

// send a message to the network
void NodeManager::sendMessage() {
	// send the message, multiple times if requested
	for (int i = 0; i < _retries; i++) {
		// if configured, sleep beetween each send
		if (_sleep_between_send > 0) sleep(_sleep_between_send);
#if DEBUG == 1
		Serial.print(F("SEND D="));
		Serial.print(_msg.destination);
		Serial.print(F(" I="));
		Serial.print(_msg.sensor);
		Serial.print(F(" C="));
		Serial.print(_msg.getCommand());
		Serial.print(F(" T="));
		Serial.print(_msg.type);
		Serial.print(F(" S="));
		Serial.print(_msg.getString());
		Serial.print(F(" I="));
		Serial.print(_msg.getInt());
		Serial.print(F(" F="));
		Serial.println(_msg.getFloat());
#endif
		send(_msg, _ack);
	}
}
```

If you are ok, I can try to prepare a PR for that. The problem is that in order to implement this modification, all sensors are to be modified (the send command, especially in onProcess method). This is not a big deal, I already did for the sensors in NodeManager source. The problem is if in your modification you add sensors, this must be done also. Same for user specific sensors.

Sorry for the long message and let me know if you want a PR.

user2684, 2017-09-30:
Love the idea! Using a single MyMesssage object sounds great. I gave up a couple of ideas in the past because requiring yet an additional message and I didn't want to eat up additional space so this will be helpful also for future enhancements. If whenever you have time you can create a PR for this would be great. I see no issue in adding a note in the v1.7 release notes to tell the users custom sensors have to take this into consideration when upgrading. Also, this should have little or no impact on the new sensors<->child ids data structure

Thanks!

user2684, 2017-12-25:
Reviewed architecture and optimized the code with #229

user2684, 2017-12-29:
Reopening (https://forum.mysensors.org/topic/6534/nodemanager/138)

user2684, 2018-01-01:
Added DISABLE_* in #251, closing

closed 2018-01-01 by user2684