mysensors/NodeManager #211: SI7021 Sensor

fabix68, 2017-08-29:
Hi, is it possible to add support for si7021 sensor  in the next version? 
tnk for your work

dakipro, 2017-09-01:
I was just about to ask for this sensor :)
Looking forward to it!

user2684, 2017-09-02:
Hi, thanks for raising this! If I'm not wrong, the SI7021 *should* work with the same SHT21 library. May you please connect the SI7021, try enabling MODULE_SHT21 and adding a SENSOR_SHT21 sensor to nodemanager? If i would work I'll make it an alias like I did for SENSOR_HTU21D.
Thanks

dakipro, 2017-09-05:
Thanks, I am waiting for more parts and boards to arrive, I will test and confirm (but can take some time :( )

abdusahin, 2017-09-23:
any update for this ?

user2684, 2017-09-24:
Strange, I have a few of those https://www.aliexpress.com/item/Humidity-Sensor-with-I2C-Interface-Si7021-for-Arduino-Industrial-High-Precision/32687207471.html and SENSOR_SHT21/SENSOR_HTU21D did work for me. Which module are you using? Do you have a link of the arduino library which is working for your module?
Thanks

abdusahin, 2017-09-25:
@user2684 When I use "SparkFun_Si7021_Breakout_Library.h" it works fine. I can get temperature and humidity measurements without any issue, however the recommended method here didn't work.

user2684, 2017-09-30:
Thanks. Having a look at the library you are using, it is definitely different from the SHT21's. This means we need to implement a brand new sensor. I'll add it to the v1.7 backlog then. Thanks!

PR #278 (2018-02-11): Added support for SI7021
Fixes #211
closed 2018-02-11 by user2684