mysensors/NodeManager #214: Request memory allocation

Akubi, 2017-09-11:
I'm maybe wrong but in constructor of Request class the value part of the message is copied using _value as destination. But I don’t see any initialization of this pointer.

Can you  please check ?
```

Request::Request(const char* string) {
	char str[10];
	char* ptr;
	strcpy(str, string);
	// tokenize the string and split function from value
	strtok_r(str, ",", &ptr);
	_function = atoi(str);
	strcpy(_value, `ptr);
```

Akubi, 2017-09-12:
May I suggest a different implementation :
```

class Request {
  public:
    Request(const char* string);
    // return the parsed function
    int getFunction();
    // return the value as an int
    int getValueInt();
    // return the value as a float
    float getValueFloat();
    // return the value as a string
    char* getValueString();
   private:
    NodeManager* _node_manager;
    int _function;
    // Size of buffer to prevent overrun 
    char _value[MAX_PAYLOAD+1];
};

Request::Request(const char* string) {
	char* ptr;
	// copy to working area
	strcpy((char*)&_value, string);
	// tokenize the string and split function from value
	strtok_r(_value, ",", &ptr);
	// get function code
	_function = atoi(_value);
	// move user data to working area
	strcpy(_value, ptr);
#if DEBUG == 1
	Serial.print(F("REQ F="));
	Serial.print(getFunction());
	Serial.print(F(" I="));
	Serial.print(getValueInt());
	Serial.print(F(" F="));
	Serial.print(getValueFloat());
	Serial.print(F(" S="));
	Serial.println(getValueString());
#endif
}

```

If ok I can prepare a pull request

user2684, 2017-09-16:
Hi, thanks for looking into this. I think you're right, the pointer isn't initialized. If you can submit a pull request on the development branch with this change would be fantastic! Thanks again

Akubi, 2017-09-16:
Hi,
No problem, I will prepare a pull request tomorrow.

closed 2017-10-01 by user2684