mysensors/NodeManager #219: ds18b20 and negative temperature values

fabix68, 2017-09-18:
I'm having problems with negative temperatures on a node with 2 ds18b20
if both sensors measure the positive temperature, the values ​​are correctly sent. If one of the 2 negative negative values ​​is ignored.
everything works correctly using the example in the library.

fabix68, 2017-09-19:
The error is in the NodeManager.cpp file in the section below here calculates the average of the detected values.

Exactly at this point
// process a float value
  else if (_value_type == TYPE_FLOAT **&& total > -1**)

user2684, 2017-09-24:
Thanks. This is then a bigger issue then I thought since total is always initialized to -1 and in a few sections of the code this same logic is used. Fixing this requires initializing to a very low negative value and change all the comparisons. Good finding thanks.

closed 2017-12-25 by user2684