mysensors/NodeManager #223: Crash on startup with latest 2.2.0-beta

user2684, 2017-09-24:
After the MySensors logo appears, the node crashes.
When MY_DEBUG is defined, the crash doesn't happen

SergioRius, 2017-10-01:
I'm afraid that issue 223 is not a NodeManager failure. I can reproduce it in a new sketch without using NodeManager.

user2684, 2017-10-01:
Closing this issue since not related to NodeManager. Thanks for testing this out!

SergioRius, 2017-10-02:
Unfortunately, seems that this problem is related to nodemanager. What initially confused me, is that after reproducing it with NM, I then tested a clean MYS sketch and also happened.
But it seems like one of the first times I tested, it also mangled something on the gateway/controller for the ID of the testing node. So after that it was failing anytime. I fixed that by giving it a ID and then setting it back to auto. MYS alone worked as expected in all versions, but as soon as I added NM, it crashed.

However I've found that defining MY_SPLASH_SCREEN_DISABLED makes it work on MYS 2.2.0-beta

> 0 MCO:BGN:INIT NODE,CP=RNNNA---,VER=2.2.0-beta
4 MCO:BGN:BFR
NodeManager v1.7-dev
LIB V=2.2.0-beta R=- E=- T=N A=A S=- B=-
INT P=3 M=255
INT P=2 M=255
7 TSM:INIT
14 TSF:WUR:MS=0
20 TSM:INIT:TSP OK
22 TSF:SID:OK,ID=69
24 TSM:FPAR
60 TSF:MSG:SEND,69-69-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
2068 !TSM:FPAR:NO REPLY
2070 TSM:FPAR
2106 TSF:MSG:SEND,69-69-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
2208 TSF:MSG:READ,0-0-69,s=255,c=3,t=8,pt=1,l=1,sg=0:0
2213 TSF:MSG:FPAR OK,ID=0,D=1
4114 TSM:FPAR:OK
4115 TSM:ID
4116 TSM:ID:OK
4118 TSM:UPL
4121 TSF:MSG:SEND,69-69-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=OK:1
4128 TSF:MSG:READ,0-0-69,s=255,c=3,t=25,pt=1,l=1,sg=0:1
4133 TSF:MSG:PONG RECV,HP=1
4135 TSM:UPL:OK
4137 TSM:READY:ID=69,PAR=0,DIS=1
4144 TSF:MSG:SEND,69-69-0-0,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
4151 TSF:MSG:READ,0-0-69,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
4163 TSF:MSG:SEND,69-69-0-0,s=255,c=0,t=17,pt=0,l=10,sg=0,ft=0,st=OK:2.2.0-beta
4174 TSF:MSG:SEND,69-69-0-0,s=255,c=3,t=6,pt=1,l=1,sg=0,ft=0,st=OK:0
4397 TSF:MSG:READ,0-0-69,s=255,c=3,t=6,pt=0,l=1,sg=0:M
<= presentation
4407 TSF:MSG:SEND,69-69-0-0,s=255,c=3,t=11,pt=0,l=11,sg=0,ft=0,st=OK:Development
4418 TSF:MSG:SEND,69-69-0-0,s=255,c=3,t=12,pt=0,l=3,sg=0,ft=0,st=OK:1.0
RADIO OK
4430 TSF:MSG:SEND,69-69-0-0,s=255,c=3,t=11,pt=0,l=11,sg=0,ft=0,st=OK:NodeManager
4442 TSF:MSG:SEND,69-69-0-0,s=255,c=3,t=12,pt=0,l=3,sg=0,ft=0,st=OK:1.0
PRES I=200, T=23
4450 TSF:MSG:SEND,69-69-0-0,s=200,c=0,t=23,pt=0,l=0,sg=0,ft=0,st=OK:
PRES I=201, T=30
4461 TSF:MSG:SEND,69-69-0-0,s=201,c=0,t=30,pt=0,l=0,sg=0,ft=0,st=OK:
BATT V=5.16 P=100
SEND D=0 I=201 C=0 T=38 S= I=0 F=5.16
4538 !MCO:SND:NODE NOT REG
4548 TSF:MSG:SEND,69-69-0-0,s=255,c=3,t=0,pt=1,l=1,sg=0,ft=0,st=OK:100
READY
4554 MCO:REG:REQ
4558 TSF:MSG:SEND,69-69-0-0,s=255,c=3,t=26,pt=1,l=1,sg=0,ft=0,st=OK:2
4566 TSF:MSG:READ,0-0-69,s=255,c=3,t=27,pt=1,l=1,sg=0:1
4571 MCO:PIM:NODE REG=1
4573 MCO:BGN:STP
<= setup
MY I=69 M=1
4575 MCO:BGN:INIT OK,TSP=1
<= loop

Also without MY_DEBUG:

> NodeManager v1.7-dev
LIB V=2.2.0-beta R=- E=- T=N A=A S=- B=-
INT P=3 M=255
INT P=2 M=255
<= presentation
RADIO OK
PRES I=200, T=23
PRES I=201, T=30
BATT V=5.16 P=100
SEND D=0 I=201 C=0 T=38 S= I=0 F=5.16
READY
<= setup
MY I=69 M=1
<= loop
<= loop

user2684, 2017-10-08:
Thanks a lot for you tests! Defining MY_SPLASH_SCREEN_DISABLED seems way easier and with less impacts rather than enabling MY_DEBUG which by the way consumes a lot of storage. I'll keep the situation monitored and if the problem won't be solved by itself in the future beta versions, I'll set MY_SPLASH_SCREEN_DISABLED by default in NodeManager. Many thanks once again!

closed 2017-12-25 by user2684