mysensors/NodeManager #224: Power Management GND-On/Vcc-On

user2684, 2017-09-30:
https://forum.mysensors.org/topic/6534/nodemanager/109
https://forum.mysensors.org/topic/6534/nodemanager/120

> Expanding onto my earlier post on the requested Power Management feature -
> A simple switch will complete a circuit whenever its switched ON. This allows for placing the switch on the ground return wire too. So only a single GND-ON (Switched ON) will complete the circuit (without requiring an additional Vcc-ON switch).
> 
> Now, with a developer configured #DEFINE GND_ON_POWER =1 may be checked in NodeManager's PowerManagement functions to DigitalWrite(Digital_GND_ON_Pin, LOW)
> for a switchON effect.
> 
> Additionally, To provide an extra current to a sensor (i.e. more than a ~10mA-20mA, a max safe current provided by an Arduino pin) an external power source (other than Arduino +5V0 OR +3V3) may be provided by that external power source (of course with with a common ground wire between Arduino and that powersource to complete the circuit.)
> 
> Also, with the current 2Wire PowerOn methodology, the VccPowerOn switch can provide only one positive voltage (that provided by Arduino Vcc, say +5V0) but not +3V3 (say) to the sensors. If some attached sensors require the other voltage (+3V3), then, this 2Wire PowerOn method will not help. Whereas, a single GND_On power methodology will still switch the circuit thus managing the Power to those sensors too.
> [PS: Signal Voltage Levels will still need to be managed by the developer attaching those sensors to the node]
> 
> I think this simple feature will add on to the power of the PowerManagement functions.

luizrrocha, 2017-10-01:
I would never switch off the GND pin of a sensor and leave its output connected to the Arduino without an internal path to GND....Basic Electricity 101 :-)

And if you add lets say a transistor switch to the GND of a sensor, the voltage drop across the transistor can be enough to raise the 0 level (0 to 0.8V above ground usually) sensor output so that the Arduino can mistakenly interpret it as a 1.

I have been looking at NodeManager's supply control and I have been always concerned that the two voltage drops induced by using Arduino pins to control both VCC and GND can render some sensors powerless if you use for instance two AA batteries to supply the circuit. I will probably  never use the GND switching to a sensor for this reason and the one stated above.

One other thing: if you switch off the GND of a 3.3V sensor connected to a 5V Arduino, there will certainly be a strong current flowing from the Arduino pin thru the sensor output towards the sensors VCC pin. Almost all input or input/output pins have pull-up resistors or clamp diodes to VCC internally...

My 2 cents...

user2684, 2017-10-01:
Thanks for the comment. All reasonable concerns I think. I'll definitely make this as an option and not as the default behavior to be safer when I'll come to it. Thanks!