mysensors/NodeManager #230: error: 'int16_t DallasTemperature::millisToWaitForConversion(uint8_t)' is private

deisi, 2017-10-24:
Compilation error if `#define MODULE_DS18B20 1` is enabled.
 The dependencies are installed with the versions:

- Bibliothek MySensors in Version 2.1.1
- Bibliothek SPI in Version 1.0
- Bibliothek OneWire in Version 2.3.3
- Bibliothek DallasTemperature in Version 3.7.6 

I tired Downgrading OneWire and DallasTemperature to the oldest availabe in the IDE but get the same error. The full error message is:

    In file included from C:\Users\malte\AppData\Local\Temp\arduino_build_835316\sketch    \NodeManager.h:359:0 from C:\Users\malte\AppData\Local\Temp\arduino_build_835316\sketch\NodeManager.cpp:5:
    
    C:\Users\malte\Documents\Arduino\libraries\DallasTemperature/DallasTemperature.h: In member function 'virtual void SensorDs18b20::onLoop()':
    
    C:\Users\malte\Documents\Arduino\libraries\DallasTemperature/DallasTemperature.h:252:13: error: 'int16_t DallasTemperature::millisToWaitForConversion(uint8_t)' is private
    
         int16_t millisToWaitForConversion(uint8_t);
    
                 ^
    
    NodeManager.cpp:1501: error: within this context
    
         int16_t conversion_time = _sensors->millisToWaitForConversion(_sensors->getResolution());
    
                                                                                                ^
    
    Bibliothek MySensors in Version 2.1.1 im Ordner: C:\Users\malte\Documents\Arduino\libraries\MySensors  wird verwendet
    Bibliothek SPI in Version 1.0 im Ordner: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI  wird verwendet
    Bibliothek OneWire in Version 2.3.3 im Ordner: C:\Users\malte\Documents\Arduino\libraries\OneWire  wird verwendet
    Bibliothek DallasTemperature in Version 3.7.6 im Ordner: C:\Users\malte\Documents\Arduino\libraries\DallasTemperature  wird verwendet
    exit status 1
    within this context

Anything I've done wrong obviously? Its mir first node with NodeManager.

user2684, 2017-11-05:
Hi, I wonder if it is because of the library you are using. As per the instruction NodeManager uses https://github.com/milesburton/Arduino-Temperature-Control-Library when MODULE_DS18B20 is enabled. May you please check if it is due to a library issue?
Thanks!

deisi, 2017-11-06:
Okay, I check ASAP

sensorii, 2017-12-14:
Same issue encountered using version 3.7.7 of the DallasTemperature lib with platformio:
```
 lib_deps =
   DallasTemperature@3.7.7
```

Using the latest master branch of the lib fixed it for me:
```
 lib_deps =
   https://github.com/milesburton/Arduino-Temperature-Control-Library.git#1b2e9f46669a37d996cec7a3a63ce6295889c59a
```

Related issue:
https://github.com/milesburton/Arduino-Temperature-Control-Library/issues/86

deisi, 2018-03-02:
Hey I forgot about this, but I think you are right. By using https://github.com/milesburton/Arduino-Temperature-Control-Library it went away. THX

user2684, 2018-03-03:
Thanks for the feedback, it might be due to some update of the library. I'm closing this for now, please feel free to reopen it should you experience the same issue. Thanks!

closed 2018-03-03 by user2684