mysensors/NodeManager #233: [enhancement] Optimization for button cell batteries (CR2032 and the likes)

dakipro, 2017-11-19:
Hi, 
I wrote on the forum suggestion, thought I might also post here a suggestion about the optimization for button cell batteries.
I am not an expert in cell batteries, but by looking at this sketch https://www.openhardware.io/view/398/NModule-Temperature-Humidity-Light-Door-sensor-shield#tabs-source
main optimizations are done relatively simple, f.eks. short 400ms sleeping between each data sending, short sleep after presentation, and I guess a few other relevant things (maybe voltage reporting could/should be set differently).

Mentioned forum discussion here https://forum.mysensors.org/topic/7998/optimisation-for-button-cells-cr2032-and-the-likes

user2684, 2017-11-25:
Hi, thanks for this report! Personally I always had negative experiences with coin cells and arduino's boards. Even forcing the board to sleep between each individual "send" (already available in NodeManager) looks like there is no enough current to have the board consistently reporting back (in my tests it worked once out of 10 tries). But this is just my personal experience so nothing wrong to implement those optimizations and give this another try :) 
I'll add it to the v1.7 queue.
Thanks

dakipro, 2017-11-25:
Thanks. I think that a few other "hardware" tweaks are needed as well, f.eks. big cap (100-200uf) to help the battery, and a 1Mhz  bootloader with lowered BOD. I made two sensors fallowing the instructions on the linked NModule and they work great, no issues with the coin cell battery

user2684, 2017-12-02:
Good to know! I'll give it another try while implementing it in NodeManager then :)

user2684, 2018-01-20:
Done but in my latest tests I was still unable to have it working, voltage drop to 1.4v making the board unable to start (even running at 1Mhz and with BOD disable), but this is another story :)
Thanks

PR #255 (2018-01-20): Optimization for button cell batteries
Fixes #233 

* Added defines from https://www.openhardware.io/view/398/NModule-Temperature-Humidity-Light-Door-sensor-shield#tabs-source into the main sketch which must be manually uncommented when running on a cell coin (cannot be done programmatically)
* Sleeping between send is already implemented, just moved _sleepBetweenSend() after each send() and present()

To run on coin cell:
* Uncomment into the main sketch the defines under "Optimizations when running on 2032 Coin Cell"
* Set node.setSleepBetweenSend(500)
* Run the board at 1Mhz and disable BOD
closed 2018-01-20 by user2684