mysensors/NodeManager #238: Unable to compile vanilla NodeManager

fookatchu, 2017-12-21:
Hi!

If I try to compile NodeManager 1.6 without (and with) any modifications with Arduino 1.8.5 on Windows 10, I get the following compiler error:

```
sketch\NodeManager.cpp: In member function 'int NodeManager::registerSensor(Sensor*)':

NodeManager.cpp:3360: error: return-statement with no value, in function returning 'int' [-fpermissive]

   if (sensor->getChildId() > MAX_SENSORS) return;

                                           ^

sketch\NodeManager.cpp: In member function 'bool NodeManager::renameSensor(int, int)':

NodeManager.cpp:3400: error: return-statement with no value, in function returning 'bool' [-fpermissive]

   if (old_child_id > MAX_SENSORS || new_child_id > MAX_SENSORS) return;

                                                                 ^

exit status 1
return-statement with no value, in function returning 'int' [-fpermissive]
```

From my standpoint they look to be quite obvious errors but I guess something is wrong on my side, since these errors should make it impossible for anyone to compile. Am I missing something?

fookatchu, 2017-12-21:
The mentioned error gets only raised when I try to compile for a Moteino board. If I try to compile for an Arduino Leonardo board only warnings are raised. Here is a more verbose output of the compiler:

```
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Users\xxxxxxx\Documents\Arduino\libraries\MySensors" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\xxxxxxx\AppData\Local\Temp\arduino_build_325213\sketch\NodeManager.cpp" -o "C:\Users\xxxxxxx\AppData\Local\Temp\arduino_build_325213\sketch\NodeManager.cpp.o"
C:\Users\xxxxxxx\AppData\Local\Temp\arduino_build_325213\sketch\NodeManager.cpp: In member function 'int NodeManager::registerSensor(Sensor*)':

NodeManager.cpp:3360: error: return-statement with no value, in function returning 'int' [-fpermissive]

   if (sensor->getChildId() > MAX_SENSORS) return;

                                           ^

C:\Users\xxxxxxx\AppData\Local\Temp\arduino_build_325213\sketch\NodeManager.cpp: In member function 'bool NodeManager::renameSensor(int, int)':

NodeManager.cpp:3400: error: return-statement with no value, in function returning 'bool' [-fpermissive]

   if (old_child_id > MAX_SENSORS || new_child_id > MAX_SENSORS) return;

                                                                 ^

```

user2684, 2017-12-24:
Hi, depending on the board selected different compiler options are applied and -fpermissive usually hides those errors (similar issues #155 #82 #30), this is why usually does not show up. You can either add this option or a return value. Nevertheless, this has to be fixed so let me tag it as a bug :)
Thanks!

user2684, 2017-12-25:
Code now compiles fine without -fpermissive

closed 2017-12-25 by user2684