mysensors/NodeManager #254: Add option to blink a led when battery is low

user2684, 2018-01-20:
e.g. https://www.openhardware.io/view/398/NModule-Temperature-Humidity-Light-Door-sensor-shield#tabs-source

dudman8, 2018-02-25:
Or register a callback function so it can be implemented by user more flexabily ? Just idea.

user2684, 2018-02-26:
Do you have an example to share? I'm interested in this callback approach. Thanks

user2684, 2018-03-03:
Interesting thanks, I've opened #307 specifically for this. The main issue I see in NodeManager to have this to work is that the callback has to access the Sensor's variables to do something significant (e.g. retrieve the data, call another sensor's function etc.) but I'll investigate further. Thanks!

WinndyBoy, 2018-03-03:
Another possibily related point. how does a switch defined in nodemanager interract directly with a light ? Does it always have to be via a controller, or could the call back functions also be possible to allow a switch single or a value change to be handled locally at a higher level than the NM ? Of course the same problem of accessing relavent data from the sensor, but just pass that in the call back function ?

user2684, 2018-03-04:
Hi, this interaction between sensors on the same node without the need of involving the controller is for sure something we need to take care of. Not an easy task due to the nature of NodeManager which cannot cover all the custom use cases out of the box but my interest in having a callback function is exactly going into this direction. As soon as I will be able to perform some tests, I'll update #307. Thanks!

user2684, 2018-03-04:
Hi, for inter-sensor communication, have a look at PR #312 and let me know on issue #307 if this may be the first step towards a nicer solution. Thanks

user2684, 2018-03-10:
Closing this down since this can be implemented already by hooking into SensorBattery, obtaining the latest value and if too low, calling setStatus() of a SensorRelay connected to a LED.

```c
SensorBattery battery(node);
SensorRelay led(node);

// set the digital output of a pin, equals to the digital input of another pin
void batteryLow(Sensor* sensor) {
  float level = ((ChildInt*)sensor->children.get(1))->getValueFloat();
  if (level < 2) led.setStatus(ON);
}

// before
void before() {
  // setup the serial port baud rate
  Serial.begin(MY_BAUD_RATE);
  
  battery.setPostLoopFunction(&batteryLow);
  led.setPulseWidth(1000);
}
```

closed 2018-03-10 by user2684