mysensors/NodeManager #257: Add support for "virtual" sensors which are not tied to pins

jsiddall, 2018-01-24:
Add the ability to define a child sensor using NodeManager which does not connect to a physical pin but instead has software hooks that can be used by user code.

This feature should allow a user to easily define a sensor using NodeManager, as they can today, but using a callback function that NodeManager would call when a message was received instead of a physical pin.  This effectively replaces the physical pin with a function call which the user can handle appropriately.

A function to send a message to the MyS controller would also be required effectively allowing user code to replace an input pin.

user2684, 2018-01-24:
Hi, sensors not tied to pin can be implemented somehow even today (especially with the dev code base) since pin is not mandatory. E.g. classes using i2c to communicate with the sensor do not request the pin from the user. Tell me more about the use case just to be sure I got your point correctly.
Hooking into the code has been proposed also in #103. Something to evaluate for sure.
Thanks!

jsiddall, 2018-01-24:
One specific use case is a TTP229 touch sensor acting as an MyS remote control. This device has 8 or 16 buttons and sends the state changes as serial data over a single data pin. I would like to treat these button presses as inputs to S_BINARY sensors.  Ex: one button could be ON for sensor 0 and another could be the OFF for that sensor.  A separate pair of buttons could be ON and OFF for sensor 1, and so on.

To make this work I would need to check the buttons in the main loop and then, depending on what button was pressed, update the state of the appropriate sensor to trigger it to send a SET message to the controller.

Another example is to use a sensor to hold arbitrary information.  For example, create a CUSTOM_S sensor that stores a timestamp in V_VAR1, a bit like I_TIME.  This would either require a hook, similar to the receive function, so that some user code could be triggered when a message for this sensor arrived.  Even without a callback function perhaps the state of the sensor could be polled in the main loop to check for updates.  Then, when that time arrived the Arduino could do something.

If there are already ways to do this in the current dev branch then an example of that would be great.  If there is no need to write code to make this work then that is even better!

user2684, 2018-01-27:
Hi, yes, I think custom sensors is the way to go. Just create your class deriving from Sensor and then create an instance of the new class and you should be done. Regarding the first example I'm not completely sure I've understood correctly. You can still implement your logic in the custom onLoop() and theoretically achieve what you need. Have a look at the SensorSonoff class, it should have a similar logic. Let me know of course if my understanding is somehow wrong :) Thanks

user2684, 2018-03-31:
Closing this, feel free to reopen if you think has not been addressed correctly. Thanks!

closed 2018-03-31 by user2684