mysensors/NodeManager #258: Invalid messages being sent upon wake and sleep

liwenyip, 2018-01-26:
My node is sending strange messages every time it sleeps and wakes up.

The offending messages, as seen in serial monitor are:
```
4765 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=32,pt=5,l=4,sg=0,ft=0,st=OK:500
5292 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=33,pt=5,l=4,sg=0,ft=0,st=OK:9000
```

The message type is [undefined](https://www.mysensors.org/build/parser?log=4765%20TSF%3AMSG%3ASEND%2C242-242-0-0%2Cs%3D255%2Cc%3D3%2Ct%3D32%2Cpt%3D5%2Cl%3D4%2Csg%3D0%2Cft%3D0%2Cst%3DOK%3A500%0A5292%20TSF%3AMSG%3ASEND%2C242-242-0-0%2Cs%3D255%2Cc%3D3%2Ct%3D33%2Cpt%3D5%2Cl%3D4%2Csg%3D0%2Cft%3D0%2Cst%3DOK%3A9000) and I can't tell what purpose they serve. How do I disable them?

I can't find where in the sketch these messages are sent from.

This is is the full serial monitor output:

```
 
 __  __       ____
|  \/  |_   _/ ___|  ___ _ __  ___  ___  _ __ ___
| |\/| | | | \___ \ / _ \ `_ \/ __|/ _ \| `__/ __|
| |  | | |_| |___| |  __/ | | \__ \  _  | |  \__ \
|_|  |_|\__, |____/ \___|_| |_|___/\___/|_|  |___/
        |___/                      2.2.0-rc.2

18 MCO:BGN:INIT NODE,CP=RPNNA---,VER=2.2.0-rc.2
28 MCO:BGN:BFR
NodeManager v1.6
LIB V=2.2.0-rc.2 R=P E=- T=N A=A S=- B=-
INT P=3 M=255
INT P=2 M=255
32 TSM:INIT
38 TSF:WUR:MS=0
40 TSM:INIT:TSP OK
43 TSF:SID:OK,ID=242
45 TSM:FPAR
51 TSF:MSG:SEND,242-242-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
598 TSF:MSG:READ,0-0-242,s=255,c=3,t=8,pt=1,l=1,sg=0:0
604 TSF:MSG:FPAR OK,ID=0,D=1
2058 TSM:FPAR:OK
2058 TSM:ID
2060 TSM:ID:OK
2062 TSM:UPL
2070 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=OK:1
3305 TSF:MSG:READ,0-0-242,s=255,c=3,t=25,pt=1,l=1,sg=0:1
3311 TSF:MSG:PONG RECV,HP=1
3315 TSM:UPL:OK
3317 TSM:READY:ID=242,PAR=0,DIS=1
3330 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
3340 TSF:MSG:READ,0-0-242,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
3362 TSF:MSG:SEND,242-242-0-0,s=255,c=0,t=17,pt=0,l=10,sg=0,ft=0,st=OK:2.2.0-rc.2
3385 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=6,pt=1,l=1,sg=0,ft=0,st=OK:0
3411 TSF:MSG:READ,0-0-242,s=255,c=3,t=6,pt=0,l=1,sg=0:M
RADIO OK
3434 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=11,pt=0,l=11,sg=0,ft=0,st=OK:NodeManager
3457 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=12,pt=0,l=3,sg=0,ft=0,st=OK:1.0
PRES I=200, T=23
3481 TSF:MSG:SEND,242-242-0-0,s=200,c=0,t=23,pt=0,l=0,sg=0,ft=0,st=OK:
PRES I=201, T=30
3504 TSF:MSG:SEND,242-242-0-0,s=201,c=0,t=30,pt=0,l=0,sg=0,ft=0,st=OK:
BATT V=3.32 P=100
SEND D=0 I=201 C=0 T=38 S= I=0 F=3.32
3584 !MCO:SND:NODE NOT REG
3598 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=0,pt=1,l=1,sg=0,ft=0,st=OK:100
READY

3606 MCO:REG:REQ
3620 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=26,pt=1,l=1,sg=0,ft=0,st=OK:2
3631 TSF:MSG:READ,0-0-242,s=255,c=3,t=27,pt=1,l=1,sg=0:1
3637 MCO:PIM:NODE REG=1
3639 MCO:BGN:STP
MY I=242 M=1
3641 MCO:BGN:INIT OK,TSP=1
SLEEP 9s

3645 MCO:SLP:MS=9000,SMS=1,I1=255,M1=255,I2=255,M2=255
3663 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=32,pt=5,l=4,sg=0,ft=0,st=OK:500
4173 TSF:TDI:TSL
4175 MCO:SLP:WUP=-1
4177 TSF:TRI:TSB
4188 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=33,pt=5,l=4,sg=0,ft=0,st=OK:9000
AWAKE
SLEEP 9s

4198 MCO:SLP:MS=9000,SMS=1,I1=255,M1=255,I2=255,M2=255
4212 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=32,pt=5,l=4,sg=0,ft=0,st=OK:500
4720 TSF:TDI:TSL
4722 MCO:SLP:WUP=-1
4724 TSF:TRI:TSB
4741 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=33,pt=5,l=4,sg=0,ft=0,st=OK:9000
AWAKE
SLEEP 9s

4751 MCO:SLP:MS=9000,SMS=1,I1=255,M1=255,I2=255,M2=255
4765 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=32,pt=5,l=4,sg=0,ft=0,st=OK:500
5271 TSF:TDI:TSL
5273 MCO:SLP:WUP=-1
5275 TSF:TRI:TSB
5292 TSF:MSG:SEND,242-242-0-0,s=255,c=3,t=33,pt=5,l=4,sg=0,ft=0,st=OK:9000
AWAKE
SLEEP 9s

```

My sketch is exactly as downloaded v1.6 (https://github.com/mysensors/NodeManager/commit/1436caf12ecc820a59dbd7e570102abf4649ae83) with the following change only:

```
  /*
   * Register below your sensors
  */
  
  nodeManager.setSleepSeconds(9);
  
  /*
   * Register above your sensors
  */
```

My config.h is exactly as downloaded v1.6 (https://github.com/mysensors/NodeManager/commit/1436caf12ecc820a59dbd7e570102abf4649ae83) with changes only to enable debug and RFM69 radio.

```

/**********************************
 * MySensors node configuration
 */

// General settings
#define MY_BAUD_RATE 115200
#define MY_DEBUG
//#define MY_NODE_ID 100
//#define MY_SMART_SLEEP_WAIT_DURATION_MS 500

// NRF24 radio settings
//#define MY_RADIO_NRF24
//#define MY_RF24_ENABLE_ENCRYPTION
//#define MY_RF24_CHANNEL 76
//#define MY_RF24_PA_LEVEL RF24_PA_HIGH
//#define MY_DEBUG_VERBOSE_RF24
//#define MY_RF24_DATARATE RF24_250KBPS

// RFM69 radio settings
#define MY_RADIO_RFM69
//#define MY_RFM69_FREQUENCY RF69_868MHZ
#define MY_RFM69_FREQUENCY RFM69_433MHZ
//#define MY_IS_RFM69HW
#define MY_RFM69_NEW_DRIVER
//#define MY_RFM69_ENABLE_ENCRYPTION
//#define MY_RFM69_NETWORKID 100
//#define MY_DEBUG_VERBOSE_RFM69
//#define MY_RF69_IRQ_PIN D1
//#define MY_RF69_IRQ_NUM MY_RF69_IRQ_PIN
//#define MY_RF69_SPI_CS D2
//#define MY_RFM69_ATC_MODE_DISABLED
```

user2684, 2018-01-27:
Hi, I think it should be "I_PRE_SLEEP_NOTIFICATION" and "I_POST_SLEEP_NOTIFICATION" that have been introduced with 2.2.0 and not related to NodeManager (https://github.com/mysensors/MySensors/blob/master/core/MyMessage.h#L197) . May you please check with a sample sketch without NodeManager if you have the same in your controller? Thanks!

liwenyip, 2018-01-29:
I think you're right, these are I_PRE_SLEEP_NOTIFICATION (t=32) and I_POST_SLEEP_NOTIFICATION (t=33), but it's very strange that they are being sent:

- They are only sent when I'm using NodeManager (I've run a sketch without NodeManager to test)
- I can't see anything in the NodeManager code that would explicitly enable these messages

I'm currently running MySensors v2.2.0-rc2. I see 2.2.0 has just come out, will update and let you know if the issue persists.

Also strange:
- The gateway, which is running the same version of MySensors, sees them as invalid messages
- Log parser doesn't recognise these message types (I've left a there comment to flag this)

liwenyip, 2018-02-02:
I found the culprit:

https://github.com/mysensors/NodeManager/blob/1436caf12ecc820a59dbd7e570102abf4649ae83/NodeManager.cpp#L4020

Uses smartSleep=true, which is what causes the I_PRE_SLEEP_NOTIFICATION and I_POST_SLEEP_NOTIFICATION messages to be sent. 

_It would be good to add an option in config.h so that smartSleep is only used if explicitly enabled, or if required by some other enabled feature that requires smartSleep to be enabled._

(Sending these messages consumes valuable battery life.)

@user2684: Happy to make the change and submit a PR if you could confirm which (if any) NodeManager features depend on smartSleep being enabled.

user2684, 2018-02-04:
Thanks, you're right, this is related to smart sleep. I've changed the title and label of this issue to turn it into an enhancement request. There is nothing which depends on smart sleep on this side so it is safe to make it as an option. Thanks

liwenyip, 2018-02-04:
No worries, I will add an option for this in config.h and submit a pull request to the dev branch.

user2684, 2018-03-03:
Fixes through #305

PR #305 (2018-03-03): Architecture consolidation
This PR is aimed to consolidate some architectural aspects:
* Updated compatibility table in documentation (fixes #261)
* Added v1.7 release notes
* Integrated NODEMANAGER_DEBUG within built-in features as FEATURE_DEBUG
* Added default FEATURE_* defines if not defined in the main sketch
* Added setSmartSleep() (fixes #258)
* Added `float_precision` to Child class (fixes #270) which is used by both ChildFloat (default to 2) and ChildDouble (default to 4). `float_precision` is then used in both NodeManager::SendMessage() and in Child::printOn(). Example on how to custo [... cut]
closed 2018-03-03 by user2684