mysensors/NodeManager #270: Float precision for temperatures should default to 1 decimal place (enhancement)

liwenyip, 2018-02-05:
None of the commonly-used temperature sensors are accurate to more than +/- 0.3degC, so a temperature reading with 2 decimal places is meaningless.

I've implemented this in my fork of the master branch by calling  `setFloatPrecision(1);` in the constructor of `SensorDS18B20` and other temperature sensors; I can't find where float precision is handled in the development branch.

liwenyip, 2018-02-10:
Yes, I think it's important to show a sensible number of decimal places for each sensor type by default. 

e.g.

- Battery voltage: 2 decimal places
- Temperature: 1 decimal place
- Humidity: 0 decimal places (or int)

This is because of the accuracy/precision of the sensors, and that more than 2 or 3 significant figures would be meaningless to most people, and can actually make the values harder to interpret. 

> I can make them global variables but they will not be per-sensor.
I wouldn't bother with this - if I want to change the precision globally, it's easy enough to change it where it's hard-coded.

user2684, 2018-02-11:
For 0 decimal places ChildInt has to be used instead. The issue is related to ChildFloat only. Let me see if I can figure out a way to achieve this without changing too much code. Thanks!

user2684, 2018-03-03:
Fixed through #305

PR #305 (2018-03-03): Architecture consolidation
This PR is aimed to consolidate some architectural aspects:
* Updated compatibility table in documentation (fixes #261)
* Added v1.7 release notes
* Integrated NODEMANAGER_DEBUG within built-in features as FEATURE_DEBUG
* Added default FEATURE_* defines if not defined in the main sketch
* Added setSmartSleep() (fixes #258)
* Added `float_precision` to Child class (fixes #270) which is used by both ChildFloat (default to 2) and ChildDouble (default to 4). `float_precision` is then used in both NodeManager::SendMessage() and in Child::printOn(). Example on how to custo [... cut]
closed 2018-03-03 by user2684