mysensors/NodeManager #271: SENSOR_SWITCH not recognised in home assistant

liwenyip, 2018-02-10:
SensorSwitch presents as S_CUSTOM with message type V_TRIPPED, which (it looks like) Home Assistant doesn't recognise.

Would it be more appropriate for SensorSwitch to present as S_BINARY/V_STATUS (which Home Assistant does recognise)?

Or is there a specific reason why it's presented as S_CUSTOM/V_TRIPPED?

user2684, 2018-02-10:
Hi, I thought S_BINARY was more for relay type of sensor whereas the SensorSwitch class is for interrupt-based sensor like a PIR but probably you are right, regardless would be better to set the defaults to something which is recognized, I'll fix it. Meanwhile you can always customize it in your own code with something like sensorSwitch.children.get(1)->presentation = something, the same for type.
Thanks!

liwenyip, 2018-02-10:
>  I thought S_BINARY was more for relay type of sensor

Actually you're right, S_BINARY appears in Home Assistant like this:

[image]

So perhaps S_CUSTOM is more appropriate, but the message type would need to be V_CUSTOM for Home Assistant (and other controllers?) to recognise it.

user2684, 2018-02-11:
Got it thanks, I'll use S_CUSTOM then. Maybe "SensorSwitch" is named in a wrong way, I'll see if it is worth renaming it entirely. Thanks!

PR #284 (2018-02-17): SensorSwitch refactor
This PR is for refactoring the SensorSwitch class.

* SensorSwitch renamed into SensorInterrupt to avoid misunderstanding (switch may sound like a relay, instead it is a general purpose interrupt based sensor)
* Dependent module USE_SWITCH renamed into USE_INTERRUPT
* SensorInterrupt now using S_CUSTOM,V_CUSTOM (fixes #271)
* The value from the pin which generated the interrupt is store just after the interrupt takes place and made available with NodeManager::getLastInterruptValue() (fixes #273)
* Adjusted SensorInterrupt to use getLastInterruptValue() instead of reading  [... cut]
closed 2018-02-17 by user2684