mysensors/NodeManager #298: NodeManager::recieve() using C_REQ

dudman8, 2018-02-28:
More a question, or misunderstanding... just wondering why c_req is used instead of c_set for set FUnctions in NodeManager::recieve() ?

I'm new to mysensors, and am trying to understand how messages are sent back->forward between nodes and gateways. And it seems like c_set is more approprate. I know its possible not important as long as the other side listens for the right command.

user2684, 2018-02-28:
Hi, this shouldn't be the case, NodeManager::receive() dispatches the inbound message to the onReceive() function of the sensor through Sensor::receive() and then it is up to each sensor to react on C_REQ/C_SET messages depending on which sensor is handling the message. Do you see a different behavior? Thanks

dudman8, 2018-02-28:
Well, you certainly know the code MUCH better than me, I'm just expressing confusion :) 
NodeManager::receive() recieves filters and redispactches messages to the appropriate places. First though I understand it checks if the incomming messages are destined for CONFIGURATION_CHILD_ID for REMOTE_configuration etc and processes them.

Then it goes on to pass them on to the individual sensors.receive() ... 

I think my question is just semantics, its seems counter intuitive to me that the message types filtered are C_REQ for all the setFUNCTION() calls, surley they should be C_SET if the purpose of the messages is to SET a value via a setFUNCTION().

Sorry for the perhaps stupid question,. But I understand you can send SET or REQ to/from nodes or gateways... I've been searching for some description of when you should use SET or REQ, I thought that REQ was used when the sender, originator of the message is querying something, expecting a reply with a value etc ?

Perhaps this question is just because I'm not following when set/req should be used. but thanks for helping me out on this.

user2684, 2018-02-28:
Your understanding is completely correct, C_SET both C_REQ can be used depending on the context. I think you are referring to the C_REQ filter for the CONFIGURATION_CHILD_ID. This is because that fake sensor is for remotely communicating with the board so just accepting requests only but this does not apply for all the other sensors which are receiving all types of messages. Closing this for now, let me know if there is anything missing. Thanks!

closed 2018-02-28 by user2684

user2684, 2018-02-28:
No questions are irrelevant ;-) Thanks to you