mysensors/NodeManager #300: Compilation errors when building NodeManager for the ESP8266

mitchmitchell, 2018-03-02:
I can build NodeManager for the Arduino with no problem, but several compilation errors are reported when attempting to build for the NodeMCU ESP8266 board:

1. There are some functions declared as returning 'int' that have some "return;" statements (e.g. returning void) -- apparently the Arduino compiler accepts this but the compiler used for the ESP8266 does not:
   a) line 3360 -- int NodeManager::registerSensor(Sensor* sensor) {
                                    if (sensor->getChildId() > MAX_SENSORS) return; <<<<<
   b) line 3400 -- bool NodeManager::renameSensor(int old_child_id, int new_child_id) {
                                  if (old_child_id > MAX_SENSORS || new_child_id > MAX_SENSORS) return; <<<<

2. On the ESP8266 apparently some constants related to the WatchDog timer and the ADC are not defined: 
    a) line 3909 and 3911 -- void NodeManager::setADCOff() {
                                          // Disable the ADC by setting the ADEN bit (bit 7) to zero
                                          ADCSRA = ADCSRA & B01111111;
                                          // Disable the analog comparator by setting the ACD bit (bit 7) to one
                                           ACSR = B10000000;
                                          }
    b)line 3775 and 3777 --     // Software reboot with watchdog timer. Enter Watchdog Configuration mode:
                                             WDTCSR |= (1<<WDCE) | (1<<WDE);
                                            // Reset enable
                                            WDTCSR= (1<<WDE);
                                            // Infinite loop until watchdog reset after 16 ms
                                           while(true){}

In other places some of these constants are protected by   #ifndef MY_GATEWAY_ESP8266 / #endif blocks, but not in the places listed.

user2684, 2018-03-03:
Hi, would you mind giving this a try? https://github.com/mysensors/NodeManager/tree/development

I'm pretty sure I fixed those issues in the development branch but better to double check :)
Thanks!

mitchmitchell, 2018-03-03:
I will try that today, stayed up all night getting it to work on an STM32 board as well so I need to rearrange some hardware to try this new version.

mitchmitchell, 2018-03-03:
I tried the new version and I get the following compilation error:

C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\NodeManager-gw-dev\NodeManagerLibrary.ino: In member function 'void NodeManager::receive(const MyMessage&)':

NodeManagerLibrary:4586: error: invalid conversion from 'const MyMessage*' to 'MyMessage*' [-fpermissive]

     sensor->receive(&message);

                             ^

NodeManagerLibrary:649: error: initializing argument 1 of 'void Sensor::receive(MyMessage*)' [-fpermissive]

 void Sensor::receive(MyMessage* message) {

      ^

exit status 1
invalid conversion from 'const MyMessage*' to 'MyMessage*' [-fpermissive]

mitchmitchell, 2018-03-03:
Casting the address of the reference to be a MyMessage* fixed the compilation error

    // call the sensor's receive()
    sensor->receive((MyMessage*) &message);

mitchmitchell, 2018-03-03:
Ok, I was able to bring up my ESP8266 gateway with this build.  Will try an Arduino and a HyTiny STM32F103TB next.

mitchmitchell, 2018-03-04:
Well it worked on the Arduino with no problems but strangely even though I had to make the same changes as I did on the older  version to get it to compile for the STM32 it wouldn't even start up once uploaded.

mitchmitchell, 2018-03-04:
Ok, this is a wierd one -- I got the new version working on the STM32 (a HyTiny board) by essentially making the same changes I did on the old versions, a few casts needed where the compiler did not like ints passed in to functions expecting enums and something about int vs uint32_t causing ambiguity in the message routines  -- the new version boots now once I disconnected the onewire sensor I had attached -- why that would make a difference between the old and new versions I have no idea but I will check it out by using a different sensor (DHT22) and then if that works I will dig into why the one wire unit doesn;t let the machine boot up completely.  If you have any thoughts, please let me know!

mitchmitchell, 2018-03-04:
OK I was able to confirm the DS18B20 works just fine on an Arduion Uno using the new NodeManager and that the DHT22 works fine on the HyTiny board using the new NodeManager so its some interaction between the code for the DS18B20 and the STM32

user2684, 2018-03-04:
Thanks for testing it out! If you can share the full list of changes you have to perform for a smooth compilation without -fpermissive, would be great so I can update the code accordingly straight way into the upcoming v1.7. Thanks!

mitchmitchell, 2018-03-04:
Will do, but it make take a day or so, doing Taxes this weekend :-(

mitchmitchell, 2018-03-07:
Ok, here are my changes;

NodeManagerLibrary:
around 709: add #if !defined(__STM32F1__)/#endif pair

#if !defined(__STM32F1__)
  if (! _battery_internal_vcc && _battery_pin > -1)
#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
    analogReference(INTERNAL1V1);
#else
    analogReference(INTERNAL);
#endif
#endif

around 4586: add cast to const MyMessage &message when passed into sensor->receive
    // call the sensor's receive()
    sensor->receive((MyMessage*) &message);
    // turn off the pin powering all the sensors

around 4625: in function void NodeManager::reboot() change  #ifndef MY_GATEWAY_ESP8266 to:
  #if !defined(MY_GATEWAY_ESP8266) && !defined(__STM32F1__)

around 4633: in function float NodeManager::getVcc() change  #ifndef MY_GATEWAY_ESP8266 to:
  #if !defined(MY_GATEWAY_ESP8266) && !defined(__STM32F1__)

around 4716:  in call to attachInterrupt add cast to last parameter
 if (_status != SLEEP) attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN_1), _onInterrupt_1, (ExtIntTriggerMode) _interrupt_1_mode);

around 4724:  in call to attachInterrupt add cast to last parameter
 if (_status != SLEEP) attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN_1), _onInterrupt_1, (ExtIntTriggerMode) _interrupt_1_mode);

around 4781: in function void NodeManager::setADCOff() change #ifndef MY_GATEWAY_ESP8266 to:
  #if !defined(MY_GATEWAY_ESP8266) && !defined(__STM32F1__)

around 4840: in function void NodeManager::sendMessage add cast to call to _message.set
  _message.set((uint32_t)value);

I know there are watchdog timers on both the ESP8266 and STM32 boards but I haven't had time to dig into them.   I don't know why the new version of node manager conflicts with attaching a DS18B20 and keeps the HyTiny from booting, (the old version of node manager has no problem with it) but I have a Black Magic Probe that I will be putting to use at some point to try and find out.

BTW, you might want to use one of the official ESP8266 processor defines instead of the MY_GATEWAY_ESP8266 to protect the sections that have inapplicable constants in case someone uses an ESP8266 without enabling the gateway functionality (not sure why they would do that except perhaps to scan SSIDs).

user2684, 2018-03-07:
Many thanks, I'll integrate those changes asap!

user2684, 2018-03-10:
Hi, I've implemented the changes you suggested (and slightly reviewed the architecture so to make it easier to apply defines based on the chip used). See #317. Would you mind to give it a try?
I have an issue though by casting to ExtIntTriggerMode as you recommend since I get a `'ExtIntTriggerMode' was not declared in this scope` when compiling for a ESP8266 board. Any idea why? Thanks!

user2684, 2018-03-11:
Nevermind, just noticed ExtIntTriggerMode is a maple-specific enum so I'll not include it in the main code base. Meanwhile I've merged #317 with the other changes. Thanks!

closed 2018-03-11 by user2684

mitchmitchell, 2018-03-11:
Ha, I was just getting ready to try it – I’ll let you know if anything turns up…

mitchmitchell, 2018-03-14:
I was able to build without problems except for the ExtIntTriggerMode as you noted.  I'm curious why a maple specific enum is specified as an argument type in a function called in non-maple specific code but I have not had time to investigate.

mitchmitchell, 2018-03-14:
LOL, I just realized what I said there -- I was building for an STM32 target when I see that one -- *nevermind* :-)

user2684, 2018-03-15:
Thanks for confirming building was successful!

jslove, 2018-04-08:
This change causes the SIGNAL report on my pro mini's to send a massive number, like 45678987. Changing the cast to (int) fixes it.

around 4840: in function void NodeManager::sendMessage add cast to call to _message.set
_message.set((uint32_t)value);

user2684, 2018-04-08:
Adding #345 to investigate this, thanks for the feedback!

PR #317 (2018-03-11): Fix compilation errors for non-AVR chips
This PR fixes compilation issues with some platforms:
* Added global CHIP_TINYX4, CHIP_TINYX5, CHIP_MEGA, CHIP_STM32, CHIP_ESP, CHIP_AVR defines
* Added USE_SIGNAL which enables MY_SIGNAL_REPORT_ENABLED and SensorSignal
* Added USE_BATTERY which enables SensorBattery
* Added USE_CONFIGURATION which enables SensorConfiguration
* Fixed compilation errors (fixes #300)