mysensors/NodeManager #301: Request function attempts to copy string into a pointer that has not been initialized

mitchmitchell, 2018-03-03:
In the function Request inside of NodeManager, there is a strcpy that appears to be attempting to copy a string using an uninitialized pointer:

Request::Request(const char* string) {
  char str[10];
  char* ptr;
  strcpy(str,string);
  // tokenize the string and split function from value
  strtok_r(str,",",&ptr);
  _function = atoi(str);
  strcpy(_value,ptr); <<<<<<<<<<<<<<<<<<<<<<<<<<<<

At this point, _value seems to be uninitialized.

mitchmitchell, 2018-03-03:
I think the way to deal with this is to move str[10] buffer from an automatic allocated on the stack of the constructor call to a private member of the class and then simply assigning the ptr to _value -- this insures that the buffer pointed to by _value sticks around after the constructor has completed but will be de-allocated when the Request object is destroyed.

user2684, 2018-03-03:
Hi, this should have been fixed with v1.7-dev, have a look at https://github.com/mysensors/NodeManager/blob/development/NodeManagerLibrary.ino#L182. Please re-open if in case you still see something wrong. Many Thanks!

closed 2018-03-03 by user2684