mysensors/NodeManager #309: 1.7 add commented #define for rf69 frequency 433

WinndyBoy, 2018-03-04:
In the now commented example. Just wondering why you changed from config.h to inlineing everything into the sketch ?

user2684, 2018-03-04:
Hi, everything from config.h has been moved into the main sketch to simplify the configuration since having it split between config.h and the main sketch itself was not ideal. Now everything is into a single place, also easier to copy paste and share. Regarding your point, RFM69 directives are all commented out, likewise v1.6 so no difference here. Is my understanding correct? Thanks

WinndyBoy, 2018-03-04:
I just mean to add a line with a different freq 433, so it can be uncommented, nothing important really, just for completness

user2684, 2018-03-08:
Sure, this can be done easily

WinndyBoy, 2018-03-08:
:)

user2684, 2018-03-10:
Added through #312  `//#define MY_RFM69_FREQUENCY RFM69_433MHZ` to the main sketch and removed any reference to RFM69_868MHZ since RFM69_868MHZ is the default value

closed 2018-03-10 by user2684