mysensors/NodeManager #310: Idea for local sensors to comunicate with each other

WinndyBoy, 2018-03-04:
Perhaps this can be done another way, or is already possble, I could see another way currently so am suggesting something like this

In the main program you'd do something like this to connect a dimmer to rotary encoder.

dimmer.registerSensorToBeNotifier(rotaryEncoderWithPushButton);

some changes to The sensor class
`
Class Sensor {
  Sensor *_registeredSensorToBeNotified = null;

  void Sensor::registerSensorToBeNotifier(Sensor *sensor);
  void Sensor::notifyRegisteredSensor(MyMessage msg);
};

void Sensor::registerSensorToBeNotifier(Sensor *sensor) {
  _registeredSensorToBeNotified = sensor;  
}

// This sensor 
void Sensor::notifyRegisteredSensor(MyMessage msg)
{
	if (_registeredSensorToBeNotified!=0) sensor->onRecieve(msg);
}`

Then in the Sensor implementation you just call where required the notifyRegisteredSensor.

just an idea

user2684, 2018-03-04:
Thanks for sharing this idea, let me first of all link this with #307 since they are trying to solve the same problem in a different way.

user2684, 2018-03-04:
After reviewing this idea I think #307 approach is a bit more flexible because:
* Custom functions can be used to notify/work with more than one sensor;
* The MyMessage structure is using a lot of storage so using it for inter-sensor communication would be a bit inefficient; 

Let me know of course if #312 would still miss some key requirements. Thanks!

WinndyBoy, 2018-03-05:
I've written a custom Sensor for EncoderwithPushButton, and am trying out the implementation I suggested at the top of this thread. See zip.zip. I'm having trouble understand how to use the 2 Child types which are added in the DimmerSensor. DimmerSensor is register with encoder.registerSensorToBeNotified(dimmer). Inside the SensorEncoder, when buttons are pushed etc SensorRotaryEncoderWithPushButton::onLoop I don't know how to get the right Type from either of the Child's that are registered inside the DimmerSensor, so that they the correct messages can be passed locally to the dimmer sensor. Can you help a little with that, I really like the new Child structure, but am just a bit confused :)
[zip.zip](https://github.com/mysensors/NodeManager/files/1780571/zip.zip)

my motivation for all this is so with one function call encoder.registerSensorToBeNotified(dimmer). two local sensors/switch type things can be connected with out any other logic but just rely upon the existing onReceive().

WinndyBoy, 2018-03-05:
I've got it working can and share they zip just for interest, or comments.
[zip0.2.zip](https://github.com/mysensors/NodeManager/files/1782419/zip0.2.zip)

user2684, 2018-03-08:
First of all, great piece of code, thanks for sharing. I think the registerSensorToBeNotified() concept is a good idea but I still have a few concerns. First of all it looks like you need the user to add some custom code in the sensor to do something with the callback which is ok but I'd like to avoid new users to get into it (advanced users can do whatever they want of course like you did). 
Also, the call to `new MyMessage()` would allocate a heavy data structure which would eat up precious storage and we have many users complaining about NodeManager consuming too much memory. 
I wonder if you can do the same with the callback recently implemented. Hooking up after loop in your encoder and then setting the dimmer accordingly. Once again, nothing against this approach, I just need to find one and only one way which would be flexible enough to fit most of the use cases :) Thanks!

WinndyBoy, 2018-03-08:
Yep agreed, I've moved away from this, but after I started it I wanted to get it working so I understood how ... just funny like that... anyway I was going to use the new callback methods to implement the same, then I realized that I could just extend the new SensorEncoder class to understand about dimmers...  see the zip for an idea, what I tried. It still seemed easier for this specific case to extend the Sensors classes and hook straight into the virtual functions than write into the the new hook function. But perhaps if I hadn't written a new SensorEncoder this would be different.

[classextension-0.1.zip](https://github.com/mysensors/NodeManager/files/1794352/classextension-0.1.zip)

user2684, 2018-03-10:
I like the idea to extend an existing class! Looks simpler and easier. I'll then keep #307 as the main way to have existing sensors to interact with each other, closing this issue, at least for v1.7. I've added a note on PR #312 for this particular use cases you have proposed. Thanks again for your precious comments!

closed 2018-03-10 by user2684