mysensors/NodeManager #311: v1.7 theres a problem with onLoop of sensors

WinndyBoy, 2018-03-04:
Which dosn't seem to be getting called. I've implemented my own Sensor and just couldn't work out why the onLoop() was only beeing called twice (I have 2 children). So I put a println(loop) in the SensorDimmer::onLoop() function... and even though I have a dimmer object un commented still the onLoop() dosn't get called.

WinndyBoy, 2018-03-04:
theres something wrong with 
`
if (! _report_timer->isOver() && ! first_run) return;`

inside the Sensor::loop()... I commented it out and my sensors onLoops are working again.

user2684, 2018-03-04:
Hi, can you please share the entire code so I can run a deeper test and check this out?
Thanks!

WinndyBoy, 2018-03-04:
Heres... I've commented a few changes I'd made to the .h and ino of the library... but basicly commenting or uncommenting line 608 of NodeManagerLibrary.ino starts/stops the SensorDimmer:onLoop() where I've put A serial.println() statement... work or not work.

[zip.zip](https://github.com/mysensors/NodeManager/files/1778770/zip.zip)

user2684, 2018-03-08:
Every sensor has a timer, when the timer is over (by default 10 minutes or can be set by setReportIntervalSeconds(), setReportIntervalMinutes(), etc.) than onLoop() is called. This is designed for sensors which have to report periodically something (e.g. a temperature). The SensorDimmer, theoretically has no reason to go into onLoop() since the sensor is controlled via onReceive(). If you need to put something into onLoop() for whatever reason, then you have to change the reporting interval to make it triggering Does it make sense? Thanks

WinndyBoy, 2018-03-08:
Ahh, thanks for that comment. I'd started to use the timers also, but hadn't noticed this. I didn't want the SensorDimmer to be updating the the ChildInt at every change of the setPercentage i.e as the encoder or held down button was scrolling the light up/down and create a message for every 1% change.

I extended the SensorDimmer, as a test, with a timer so 5 seconds after the percentage value stops changing then a message is changed. This might be a feature to add #ifdef like the others directly to the child class, but perhaps the SensorDimer with its light up/down is a special case.

user2684, 2018-03-10:
Clear, thanks, I'll close this for now since not a bug to fix. Thanks!

closed 2018-03-10 by user2684