mysensors/NodeManager #313: Prevent sending values if update too frequent

WinndyBoy, 2018-03-06:
I'm updating the SensorDimmer->setPercentage() when a button is held, and currently a message is being sent to the network on every change.

I suggest enhancing the Child::sendValue some how so that if values change quicker than say 5 seconds (configurable) then these messages arn't sent until things have settled. Perhaps the SensorDimmer is a special case and sending messages can be prevented somewhere else, but though I'd mention it for feedback.

user2684, 2018-03-10:
I think it makes sense but I'd not touch Child; probably I'd add a sort of debounce at the specific sensor which may suffer from this. Thanks!

WinndyBoy, 2018-03-10:
Thanks for the confirmation, thats what I've done with the SensorDimmer and probably I'll take my changes out and put them into an extended class. Can share later what I did. By sort of debounce do you mean a timer which waits until no changes are made for say 5 seconds and only then writes the value to the ChildInt or other ? This is what I did... or is there another simpler way.

user2684, 2018-03-11:
Yes, something like setInterruptDebounce() from #322 which is basically ignoring further interrupts for a given timeframe