mysensors/NodeManager #316: Add support for push button

user2684, 2018-03-10:


closed 2018-03-10 by user2684

Pelado-Mat, 2020-05-20:
I need only a PUSHBUTTON and using an Sensor Interrupt is not working for me:
- The bounce2 library does a better job 
- The Child presentation must be V_STATUS and S_BINARY for HomeAssistant to work correctly.

Is there any chance to support this use case?

user2684, 2020-05-24:
Hi, as for customizing the behaviour you can try with the following:
 
```
interrupt.children.get(1)-> setType(V_STATUS);
interrupt.children.get(1)-> setPresentation(S_BINARY);
```

As for bounce2, I'm sure it would do a better job but it requires storage and would have an overall impact also outside SensorInterrupt (which relies on NodeManager's interrupt handling capabilities). Which issues are you experiencing with the native debounce? Thanks!