mysensors/NodeManager #324: Revert a Relay

deisi, 2018-03-12:
Hey is it possible to revert a relay? I have a realy, that is on if LOW is set and off if HIGH is set. `setOnValue` doesn't fulfill my needs, because this changes the voltage at the pins and then everything is wrong just the other way around.
I think of something like an `invert` setting parameter for the relay, such that. When `SetOn` is called, a LOW is set at the pin, but a `True` gets returned for the state of the relay and vice versa.

Does this make sense? Its a bit hard to describe.

user2684, 2018-03-12:
Something like SensorInterrupt::setActiveState()? (https://github.com/mysensors/NodeManager/blob/development/NodeManagerLibrary.ino#L1632)
Thanks

deisi, 2018-03-12:
Oh Year that looks like it I guess. So all I would need to add this would be to add this to a relay and I can make a PR out of it or?

user2684, 2018-03-12:
exactly, I'd add it to the SensorDigitalOutput class so it will be shared among all its subclasses. If you don't mind, I'll take care of the PR, there are already too many similar functions between SensorDigitalOutput  and SensorInterrupt and I'd like to use a common naming convention to avoid any confusion. Thanks!

deisi, 2018-03-12:
Of course. Thx very much. I just wanted to be help full :+1:

user2684, 2018-03-17:
Hi, I was reviewing this one and realized you are probably looking for setOnValue() which is already available for SensorRelay, isn't it? (https://github.com/mysensors/NodeManager/blob/development/NodeManagerLibrary.ino#L1323) 

If so, I'd need anyhow to put some order in the naming convention used which is a bit messy:
* SensorInterrupt:
```
    // [101] set the interrupt mode. Can be CHANGE, RISING, FALLING (default: CHANGE)
    void setMode(int value);
    // [103] time to wait in milliseconds after a change is detected to allow the signal to be restored to its normal value (default: 0)
    void setTriggerTime(int value);
    // [104] Set initial value on the interrupt pin (default: HIGH)
    void setInitial(int value);
    // [105] Set active state (default: HIGH) 
    void setActiveState(int value);
```
* SensorDigitalOutput and subclasses:
```
	// [103] define which value to set to the output when set to on (default: HIGH)
    void setOnValue(int value);
```
* SensorPulseMeter and subclasses:
```
    // set initial value - internal pull up (default: HIGH)
    void setInitialValue(int value);
    // set the interrupt mode to attach to (default: FALLING)
    void setInterruptMode(int value);
```

user2684, 2018-03-17:
Hold on I see where the misunderstanding is/was. setOnValue() is what you are looking for which is basically inverting the value to write. However, when you communicate with the relay, the value you send is not the value to write but the status. So if you send 1, this means ON and Nodemanager will send HIGH or LOW depending on the onValue which is set so it will respond in the correct way. I'm cleaning up this a bit with #330 but without changing the logic which should be already fine. Let me know if I'm missing something. Thanks!

deisi, 2018-03-18:
I think you understood the problem :)

user2684, 2018-03-24:
Fixed by #330

closed 2018-03-24 by user2684