mysensors/NodeManager #328: SensorDs18b20::SensorDs18b20 does not call   children.allocateBlocks();

mitchmitchell, 2018-03-16:
I'm not sure if this is the cause, but the DS180B20 code is crashing on my HyTiny board though not on my two arduino boards so this may be a red herring, but while debugging the problem I noticed the constructor never calls children.allocateBlocks() before calling     new ChildFloat(this,_node->getAvailableChildId(child_id+i),S_TEMP,V_TEMP,_getAddress(i));

I added the statement but I still get a crash but since I see this in all the other sensors I thought I'd raise it so you could check if it was needed.

mitchmitchell, 2018-03-16:
Just discovered (should have known) the buffer needs to be blank or zero filled or you will get garbage at the end of the sensor string.

user2684, 2018-03-16:
Thanks for reporting this issue. The address length is fixed so shouldn't be a problem allocating it in advance I'd say. How have you filled the buffer to avoid the garbage at the end? Whatever piece of code you can share would be helpful. Thanks!

mitchmitchell, 2018-03-16:
Here is what I did:
  char* charAddr = new char[20];
  memset(charAddr,0x20,20);

mitchmitchell, 2018-03-17:
You need to null terminate the address string as well:
around line 1783:
  }
  for (int j = 0; j < 16; j++) charAddr[j] = strAddr[j];
  **charAddr[16] = 0;**
  return charAddr;
}

mitchmitchell, 2018-03-17:
or better to change the memset statement to:

  char* charAddr = new char[20];
  **memset(charAddr,0,20);**

no real reason to blank fill, just zero fill I guess.

user2684, 2018-03-17:
Hi, would you mind giving PR #329 a try? Since the address is fixes 16 chars long, I've allocated an array 17 char length and then null terminate it. What do you think about it? thanks!

mitchmitchell, 2018-03-17:
It works!   No problem with those changes.

user2684, 2018-03-18:
Great thanks for the confirmation, I'm then going to merge the PR. Thanks!

PR #329 (2018-03-18): Minor fixes for ds18b20 sensors
This PR fixes:
* Compiler warning in _getAddress() (fixes #327)
* Attempt to write into protected string constant on STM32 (fixes #328)
closed 2018-03-18 by user2684