mysensors/NodeManager #341: Sensor with multiple values

deisi, 2018-03-28:
Hey I'm trying to add an [sds011](http://aqicn.org/sensor/sds011/de/) sensor using [lib](https://github.com/ricki-z/SDS011) Problem is that it opens a serial connection and reads of two values at the same time. One for 2.5 and 10 micrometer particles. Nodemanagers Sensor classes allow only one value afaik, how should I deal with? Write a wrapper class to buffer the value?

deisi, 2018-03-28:
Just as a reference, I started on an old commit and once it works I would port to the lastest commit, but my implementation currently looks something like:

``` cpp
SensorSDS011::SensorSDS011(NodeManager* node_manager, int child_id, int pin, int sensor_type): Sensor(node_manager, child_id, pin){
  _sensor_type = sensor_type;
  setPresentation(S_DUST);
  setType(V_LEVEL);
  setValueType(TYPE_FLOAT);
  setRxTx(pin, pin+1);
}

void SensorSDS011::setRxTx(int rxpin, int txpin){
  _rxpin = rxpin;
  _txpin = txpin;
}

void SensorSDS011::onBefore(){
}

void SensorSDS011::onSetup(){
  _sds = new SDS011();
  _sds->begin(_rxpin, _txpin);
  delay(2000);
  _sds->wakeup();
}

void SensorSDS011::onLoop(){
  float p10,p25;
  int error;
  if ( _sensor_type == SensorSDS011::p10 ){
    //_sds->wakeup();
    delay(4000);
  }
  if ( _sensor_type == SensorSDS011::p25 ){
    delay(10000);
  }
	error = _sds->read(&p25,&p10);
  #if DEBUG == 1
    Serial.println("Read Values from Feinstaubsensor:");
    if (! error) {
      Serial.print("P2.5: ");
      Serial.println(p25);
      Serial.print("P10:  ");
      Serial.println(p10);
    }
    else{
      Serial.println('Read Error');
    }
  #endif
  // Store the value
  //p10 Sensor
  if (_sensor_type == SensorSDS011::p10){
    if (! isnan(p10)) _value_float = p10;
  }
  //p25 sensor
  else if(_sensor_type == SensorSDS011::p25){
    if (! isnan(p25)) _value_float = p25;
  }

  //_sds->sleep();
}

void SensorSDS011::onReceive(const MyMessage & message){
}

void SensorSDS011::onProcess(Request & request){
}

void SensorSDS011::onInterrupt(){
}

````

user2684, 2018-03-29:
Hi, with the new v1.7 architecture this should be easier. If I understood correctly, you may need a sensor with two children, sharing some data which is something you can do with v1.7. Then you can pull out the data from the sensor once or twice and send it independently through the two children. Is it something which may work in your case? Thanks

deisi, 2018-03-29:
Ah okay, sound easier. Then I take a look at 1.7 right away Thx. I would like to keep the issue open for a little longer until I know what I have to do. I will close it then.

closed 2018-03-30 by deisi